bigtreetech/BIGTREETECH-TouchScreenFirmware #1503: [FR][BUG] M117 messages are not showing if used in custom gcode

GenaUser, 2021-01-11:
M117 messages are not displayed on the screen if they are sent by a custom gcode.
On the other hand, they are shown if you use the screen terminal.
There is always a doubt about taking into account a selection on a touch screen. Being able to display a message confirming a gcode execution is important. 
For example, to confirm the TMC settings activation
(ex: M569 S1\ n M117 stealthChop activated)

oldman4U, 2021-01-11:
In this case the search function works

Please see #1050

Thank you

oldman4U, 2021-01-12:
Have you checked the ticket I mentioned above?

oldman4U, 2021-01-16:
It would be great if you could answer to your own ticket.

Thank you

3dfixer, 2021-01-20:
Configuration_adv.h

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
#define HOST_PROMPT_SUPPORT
#endif

closed 2021-03-30 by stale[bot]