bigtreetech/BIGTREETECH-TouchScreenFirmware #1504: [FR] Access to Smart Filament Detection settings in screen settings

GenaUser, 2021-01-11:
Would it be possible to allow the modification of the smart filament detector parameters in the machine settings on the screen?
These settings may need to be changed depending on the material specifications or printing type.
Downloading the firmware each time is not functional, especially since it is not possible during printing.

oldman4U, 2021-01-11:
You can change the settings by changing the config.ini using a text editor.

If you want, I can add this feature request to the FR list

sarvenn, 2021-01-12:
@oldman4U could please add invert filament run out logic option in the feature menu as well?
Mine was wrong and I changed it from config.ini file successfully. However these kind of features would be very good to have in the screen as standalone.

GenaUser, 2021-01-12:
Thanks @oldman4U, you can add this feature to your FR list.
The important for this functionality is allowed access to the parameters directly on the screen, and if possible that they can be modified during printing.

The proposition of @sarvenn is relevant, I agree.

oldman4U, 2021-01-12:
Done. Hope this will attract a developer soon.

PR #1563 (2021-01-30): Added new param in FeatureSettings menu + toggle ACK in Terminal menu…
**NEW FEATURES:**
1) Added in FeatureSettings menu the enumerated parameters defined in config.ini such as:
 * enabled, disabled
 * enabled, disabled, auto
 * etc...
 The purpose of this addition is to allow to set some settings without the need to use so frequently the config.ini.
2) Added in Terminal menu a button to toggles the terminal ACK. The changed value is not made permanent on flash (it's meant just as a temporary setting change) . This means that in order to make the change permanent on falsh the usual FeatureSet [... cut]
closed 2021-01-30 by bigtreetech