bigtreetech/BIGTREETECH-TouchScreenFirmware #1505: [FR] Do not display the « on » choice when the smart filament detection is used

GenaUser, 2021-01-11:
Could it be possible that the “on” button does not appear in the settings menu when the smart filament detector is used in the printer?
When printing, if you want to activate or deactivate the smart sensor, you must select between two "on". This suspends the printing, because there is no end filament sensor if the smart is used.

oldman4U, 2021-01-12:
Sorry, I do not use such a sensor so maybe I am wrong, but this sounds more like a bug to me!?

guruathwal, 2021-01-13:
The runout sensor settings in the TFT are for connecting the sensor to the TFT itself. if the sensor is connected to the mainboard then this settings should be set to off.

oldman4U, 2021-01-14:
I believe there is still not clear what exactly you would like to achieve. Could you please explain a bit more in detail. Maybe some pictures could help

Thank you

GenaUser, 2021-01-22:
To your TFT display, you can connect a "filament detection module" or a "smart filament detection module" (not the both).
In "Filament sensor" of the TFT screen "Feature" menu, you have 3 possibilities: "OFF" "ON" "SMART".
[image]
"OFF" deactivates the detection of one or the other
"ON" activates the "Filament detection module" (analyzes whether the signal is activated or not)
"SMART" activates the "Intelligent Filament Detection Module" (analyzes a pulse frequency).
The problem lies in the fact that these three choices are alternated with each click => OFF, ON, SMART, OFF, ON, SMART ...
If, for example, you want to activate "SMART" while printing, you must first go to "ON". This function expects a constant signal (open or closed depending on the settings), whereas if you have a smart sensor, the signal is alternated. This inevitably causes printing to stop due to filament lack.
In short, you cannot activate a "SMART filament sensor" without stopping printing.
For my printer, I removed the "ON" option by changing the code, but automatic selection based on sensor choice would be really appreciated. For example, when "fil_runout_distance:" is enabled in config.ini

killrob, 2021-01-25:
i have a problem with this sensor, it doesn't work connected to TFT35 or TFT70 or another TFTXX as you like, yesterday my filament runs out and i had no advice from the smart filament sensor, print thrown away. If you have some trick to share.....

GenaUser, 2021-01-25:
I am using this function with a dual extruder on TFT35 V2 and V3 without problems.
Check that the pin in Variants / pin_TFT35_V2_0.h (or pin_TFT35_V3_0.h) according to your model, is indeed the one to which you have connected the detector. 
Example for V2:
  #define FIL_RUNOUT_PIN PD11
Example for V3:
  #define FIL_RUNOUT_PIN PA15

In config.ini:
fil_runout_inverting: 1 (or 0)
fil_noise_threshold: (safety time to avoid bad detections in ms)
fil_runout_distance: ( 7mm for the Bigtreetech Smart detector)
By security, do the same in Configuration.h, because I don't know which one is used.
#define FIL_RUNOUT_INVERTING true (or false)
#define FIL_NOISE_THRESHOLD (in ms)
#define FILAMENT_RUNOUT_DISTANCE_MM 7

Important: In Marlin/Configuration_adv.h, activate #define M114_DETAIL 
This adds the extruders movement detail for so that the screen can compare with the value between each detector pulse.

Also verify that you have selected "SMART" in "Feature / Filament sensor" on your screen.

It is easy to test if it works. When starting to print, stop the filament with pliers or remove it to simulate a jam.

killrob, 2021-01-25:
> I am using this function with a dual extruder on TFT35 V2 and V3 without problems.
> Check that the pin in Variants / pin_TFT35_V2_0.h (or pin_TFT35_V3_0.h) according to your model, is indeed the one [...]

ok i will try at once

killrob, 2021-01-25:
ok i've tried, but no news, still not work, i'm using marlin 2.0.7.2 and those 2 
#define FIL_RUNOUT_INVERTING true (or false)
#define FIL_NOISE_THRESHOLD (in ms)

are not present.

in configuration.h this is the code concerning runout sensor:
`#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 7

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    #define FILAMENT_MOTION_SENSOR
  #endif
#endif`

killrob, 2021-01-25:
there is anywhere the pin.pdf file as for the motherboards of BTT?

baikal, 2021-01-27:
@killrob I do not own BTT SFS, but tried with a self made rotating encoder detector. Connected to the TFT, which runs the binary firmware from 2021-01-23 (f9c2f9d786abe046a4d34bced91dbc7d12d9e894), filament runout set to 'smart' in TFT menu. Marlin is bugfix-2.0.x, with all the things enabled as of the comments in config.ini, especially M114_DETAIL.

Works fine as long as **linear advance** (LIN_ADVANCE in Configuration_adv.h of Marlin) is **disabled entirely.**
With linear advance active, Marlin always reports `E:0` to the `M114 E` the TFT is pulling, regardless of how much filament passed. So the TFT does not expect the detector to be turning and therefor does not trigger ... bummer.

Looks like a bug in Marlin - or maybe an unavoidable (undocumented?) side effect of linear advance.

killrob, 2021-01-27:
> 
> 
> @killrob I do not own BTT SFS, but tried with a self made rotating encoder detector. Connected to the TFT, which runs the binary firmware from 2021-01-23 ([f9c2f9d](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/f9c2f9d786abe046a4d34bced91dbc7d12d9e894)), filament runout set to 'smart' in TFT features. Marlin is bugfix-2.0.x, with all the things enabled as of the comments in config.ini, especially M114_DETAIL.
> 
> Works fine as long as **linear advance** (LIN_ADVANCE in Configuration_adv.h of Marlin) is **disabled entirely.**
> With linear advance active, Marlin always reports `E:0` to the `M114 E` the TFT is pulling, regardless of how much filament passed. So the TFT does not expect the detector to be turning and therefor does not trigger ... bummer.
> 
> Looks like a bug in Marlin - or maybe an unavoidable (undocumented?) side effect of linear advance.
> 
> To check, you can see the M114 E and responses scrolling by when entering an empty command in the terminal on the TFT while running a print and having sensor set to 'smart' in TFT features.

Ok i will try also this way, just 10 mins to upload firmware and preheat printer

killrob, 2021-01-27:
@baikal ok, seems that with Marlin 2.0.7.2 this trick doesn't work. Launched a print and blocked filament with a plier.... nothings happens, printer still going on printing without filament

killrob, 2021-01-27:
@baikal ok it works now, i had missed a little particular you say....`which runs the binary firmware from 2021-01-23` this, i was trying with old firmware. Thank you pal when you come in Tuscany you have a beer paied :D

closed 2021-04-04 by stale[bot]

oldman4U, 2021-06-17:
@GenaUser Please check PR2002 and let me know. For me it works perfectly.

digant73, 2021-07-27:
@oldman4u this FR was implemented by Gurmeet's PR 2002 (just tested now on my sfs) so you can remove it from the list

PR #2002 (2021-06-19): Add filament sensor type selection in config
- Add filament sensor type selection in config to show only relevant options in settings.

Fixes: #1505