bigtreetech/BIGTREETECH-TouchScreenFirmware #1506: [FR] Clickable buttons during pring

xhejtman, 2021-01-12:
Is there a reason that during print, you cannot click on hotend temperature, bed temperature and fan and directly adjust them? It seems it does nothing for me to click on them. It would be nice to have these areas clickable and tunable. Is it broken just for me or it is something to be implemented?

oldman4U, 2021-01-12:
Hi. Guess you have a bigger display than 24 or 28 because otherwise you would know why such small elements are not clickable.

;-)

xhejtman, 2021-01-12:
I have actually TFT35, however, the hotend temperature area is the same as, e.g. "stop" button are. It is not the case for 24 or 28?

oldman4U, 2021-01-12:
The Stop button like all other clickable buttons on the TFT35 has a pixel hight of 95px. The buttons you mean have a pixel hight of 50px on the TFT35 and 36px on the TFT28/24. So this is exactly 37,9% of the hight of a clickable button on your screen.

xhejtman, 2021-01-12:
ok, fair enough. But for TFT35 and larger, I still think it would be a nice feature.

oldman4U, 2021-01-12:
I agree. Like taking care of the difference in hardware like CPU and memory, which would allow much more functions. Unfortunately this is a lot of work, like the ability to define the buttons shown in the UI, a more fancy UI with gauges and bars and many things more. So we stick with the generic approach until some developers are willing to spend many months of their life to change it. Not sure 3D printers like we use today will exist then;-)

Please be so kind and check all your open tickets and close them in case you do not need them anymore.

Thank you and happy printing

radek8, 2021-01-12:
I've already thought about it, too.
It would be practical.
In addition, 3 icons for additional functions would be released in the MORE menu.
If I could do it, I would be working on it :-)))

radek8, 2021-01-12:
> Tlačítko Stop jako všechna ostatní klikatelná tlačítka na TFT35 má výšku pixelu 95 pixelů. Tlačítka, která máte na mysli, mají výšku pixelu 50px na TFT35 a 36px na TFT28 / 24. To je tedy přesně 37,9% výšky klikatelného tlačítka na vaší obrazovce.

And why not use a stylus?

radek8, 2021-01-12:
The buttons in the terminal window are smaller and can be clicked :-)

oldman4U, 2021-01-12:
Hi radek8. If we could do it, we would do this and much more... but we can not;-)

The Terminal is used less often and therefore not so important and it has less buttons, so they are "bigger" on those small screens.

I would like to add one thing more. Even the TFT43 and 50 have a bigger display, the pixel size is even less than the one of the TFT35. This means that even we would accept a pixel size of 50 as clickable, the TFT43 and 50 would miss a feature which is available on the 35.

My personal opinion and I bought a TFT24 only to see how usable it is and for testing.... Using such a small screen I can not expect the same functionality like using a bigger screen and I was surprised how reliable I was able to touch ALL the buttons. Buying a Tesla 3 you are not expecting the same comfort and functionality compared to buying a more expansive one.

kisslorand, 2021-01-12:
Users with small displays and sausage fingers will complain against it.

radek8, 2021-01-13:
Kisslorand and wouldn't you like to implement it?
I think you could handle it ;-)

kisslorand, 2021-01-13:
@radek8 
Yes, I would like to do it but I have only one TFT that is compatible with this repository, the MKS TFT28.
I would also like to enable the full RAM capacity on BTT TFTs but I cannot do that also because I have none.
Working blindly is not my style, I have to be able to test things and stress them as much as possible to find as much bugs as possible.
That's on aspect.
The other one is... I don't know how to put it gently... just check what happened in another PR (#1466) that implemented a FR, a PR that had zero bugs, didn't remove anything, didn't interfere nor had any conflict with master or any other PR. 

Anyway thanks for the consideration.

sarvenn, 2021-01-13:
I thought about the same feature yesterday and I was just about to open a ticket for the same proposal.

radek8, 2021-01-13:
@kisslorand
I read it and it would be sad. Instead of working with Oldmane4U on a joint project, you argued about a problem that wasn't there. In fact, it was just an escalation of the dispute you had before (PR # 1359)
Try to both of them and start again. (unfortunately you are both in the sign of the ram)
You can program. Neither I nor Olaman4U can do that. It would be a shame for you to leave. Oldaman, on the other hand, has organizational skills and tries to give this project some order and to filter real FW errors from user errors.

oldman4U, 2021-01-14:
One thing more.

From what I have learned, it is a lot of work to break up the two lines, 4 buttons each line structure. Even more, that this would mean two lines 3 buttons each (maybe 4 for the TFT70) where the buttons have a different width. This means a lot of work and it is still the question how this can work on smaller screens. At the moment those elements also look different than real buttons and show values, something which is not really very common for buttons. Anyway, I add it to the FR list, maybe one day someone is willing to spend so much time for this functionality.

sarvenn, 2021-01-15:
As a user let me give my positive feedbacks upon this topic. 
It would be quite intuitive and user friendly to click on those print status screen elements and go to relevant adjustment pages. First time I used my TFT35 and printing my first object, the first thing I tried was to touch on TO, Bed, Fan boxes to go to their adjustment pages. Then I figured out that they were placed in another place, just a two click away place.
Not a big issue, actually this is not a issue but there is much more user friendly and intuitive way for a user. But, on the other hand there are lots of other FRs awaited in the list which are much more beneficial or urgent. Considering the restricted sources, I think it is not a top priority item. However, I would be happy to see a developer to take the lead and have the courage to achieve this challenging feature offer. Since I do not have any technical background about coding, I can not imagine how hard or how complicated it is, but as a real life professional specialized on Customer Concentricity and User Experience I strongly support this idea. I hope I could express my thoughts sincerely.

oldman4U, 2021-01-15:
sarvenn. Whats about smaller screens where the hight of those elements is a bit more than 1/3 of a button on the TFT35?

radek8, 2021-01-15:
I think that owners of smaller displays would also welcome this improvement. and if we had a problem using the small touch area, they would use the stylus or go to the More menu, as was the case now

sarvenn, 2021-01-15:
> sarvenn. Whats about smaller screens where the hight of those elements is a bit more than 1/3 of a button on the TFT35?

I totally understand that concern. Even I use a TFT35 I also sometime use a stylus :)
As I said I know it is challenging and smaller screen users will not get the benefit of the change offer truely, but idea is a good one :)
Cheers.

kisslorand, 2021-01-21:
I started to work on this but to make it work and to look flawlessly it is a really huge amount of work needed as that menu does not fall to the 8 icon menu standard (I spare you with the boring details). The amount of work needed and the waste of program space doesn't equals, at least in my perspective, the benefits. More than that, in my tests, on my 2.8" TFT it was very cumbersome to hit them with my fingers and I do not have large ones (fingers).
As a workaround maybe a click on any of those should bring up the "More" menu and "More" menu could be replaced with something else, like "Filament change" (M600), an option that we do not have at the moment.
Food for thoughts...

kisslorand, 2021-01-21:
Just had an idea...

What if the print menu looks like the Status menu, the upper 4 icons from the Status menu, the lower 4 as we already have in the Prining menu and to have that grey area from Status menu and have on that area the percentage, time and layer height?
That way it's a piece of cake to implement the touch of nozzle temp, bed temp, fan and speed/flow icons.
We will still have all the infos but with easy access to the most important parameters.

sarvenn, 2021-01-22:
> Just had an idea...
> 
> What if the print menu looks like the Status menu, the upper 4 icons from the Status menu, the lower 4 as we already have in the Printing menu and to have that grey area from Status menu (where X, Y and Z is displayed) and have on that area the percentage, time and layer height?
> That way it's a piece of cake to implement the touch of nozzle temp, bed temp, fan and speed/flow icons.
> We will still have all the infos but with easy access to the most important parameters.
> It also would save some SPI Flash memory space removing 6 icons from it.

Hi @kisslorand,
Will lower line buttons stay where they are with the same size? (Babystep, more, cancel button etc.) I also use thumbnail view and I don't want to use it as well. Do you have a preview of your suggestion?

As I mentioned before. This idea is a good one but not a must to have. I also thought about the same thing and I was just about to open a ticket and saw this ticket.
Let's think about that, how often do we change the nozzle and bed temperature while printing. May be at the beginning of the print if we see some adhesion issues etc. So there might be more important and beneficial FRs in the backlog.

radek8, 2021-01-23:
> Just had an idea...
> 
> What if the print menu looks like the Status menu, the upper 4 icons from the Status menu, the lower 4 as we already have in the Prining menu and to have that grey area from Status menu (where X, Y and Z is displayed) and have on that area the percentage, time and layer height?
> That way it's a piece of cake to implement the touch of nozzle temp, bed temp, fan and speed/flow icons.
> We will still have all the infos but with easy access to the most important parameters.
> It also would save some SPI Flash memory space removing 6 icons from it.

@kisslorand  this comes as a good idea to me.
Do you mean something like that?
Na horní řadu ikon by se dalo kliknout. Ušetří se  3 pozice v nabídce Více, 6 původních ikon by se dalo smazat.
I would like that.
Could others comment on this? So you don't do some work that others wouldn't accept.
Can you comment on this proposal? @bigtreetech @guruathwal @digant73  @oldman4U and more

[image]

kisslorand, 2021-01-23:
Yes, it's exactly this what I had in mind. I am still thinking to make that grey bar to be a progress bar, but that would make hard to read the text on it.

guruathwal, 2021-01-23:
This is what it was like at the beginning. The line in the centre will take almost full width with actual fonts size. With text cramped together, it will make it hard to read what is what. I settled with the current style after considering many layouts because each info has a proper emphasis & spacing and it clearly differentiates between a status screen and printing screen. The info icons were not made clickable because the resistive touch screen used on these devices are not very accurate even after calibrating with a stylus and it is very much noticeable when clicking the corner buttons in the Terminal menu with small buttons. All the option are already a click away in 'More' menu.

kisslorand, 2021-01-24:
Thanks for the enlightening @guruathwal .
I guess this settles things down, time to move on.

oldman4U, 2021-01-28:
my 2 cents.

What about adding a "During Print Shortcut" function, which allows to define what is opened (None, Heat Nozzle, Heat Bed, Fan, Flow or Speed) in case the user clicks somewhere in the upper area during a print?

So the area where we have no clickable function today would be one big clickable area using this function.

closed 2021-04-05 by stale[bot]

digant73, 2021-07-27:
@oldman4u also this FR has been implemented and could be removed from the list