bigtreetech/BIGTREETECH-TouchScreenFirmware #1512: [Q] Host Start?

doug-scott, 2021-01-14:
I am using Marlin 2.0.7.2 with a BTT TFT35 E3 V3 touch screen.  I have #define HOST_START_MENU_ITEM enabled in my configuration_adv.h and when viewing the touchscreen in Marlin mode I have a menu option for "Host Start".  According to the comments in the configuration_adv.h file the #define HOST_START_MENU_ITEM created that entry on the touchscreen.  When I click on that menu option I do not see anything happen or change.  What is supposed to happen?  Is it supposed to allow the user to create a sub menu under the Host Start menu item?

oldman4U, 2021-01-14:
@qwewer0

Would you please be so kind and help us out with this request.

Thank you

qwewer0, 2021-01-14:
> Adds remote start to host action commands. Will request the host to start the active file. Useful when repeating the same file from Octoprint, or when starting the uploaded file after prep work (eg loading correct filament) is complete without going between both screens constantly.

@oldman4U But it wasn't me who added that feature, I just made it optional. (https://github.com/MarlinFirmware/Marlin/pull/19398, https://github.com/MarlinFirmware/Marlin/pull/19443)

oldman4U, 2021-01-14:
qwewer0, I am sorry, thought you have been the one introducing it.

doug-scott. Any questions left?

doug-scott, 2021-01-14:
qwewer0 :  Thank you, not something I am ever likely to use.

oldman4U:  Nope, that met my needs.

closed 2021-01-14 by doug-scott