bigtreetech/BIGTREETECH-TouchScreenFirmware #1519: MKS TFT24 V1.1 SD Card reader doesn't work

piprom, 2021-01-15:
Hello friends, i need a help with this problem:
Today i recieved my MKS DLC V2.0 mainboards with MKS TFT24 V1.1.
I try to read a files on SD card, but the board MKS TFT24 v1.1 didn't read any files (or didn't recognize the card).
I write a *.bin file on the SD card to try the reflash the MSK TFT24 v1.1. board for the laser engraver, but the process didn't started.
I check the connection between card adapter and MSK TFT24 v1.1. and everything is OK ( I guess that may be some connections are failed, but everything is OK.)
Please, tell me what I'm doing to read the card and flash the board to turn it to off-line laser enegraver.
Thank you in advance.

oldman4U, 2021-01-16:
Hi. This repository is for the FIRMWARE of BIGTREETECH TFT screens.

Thank you

oldman4U, 2021-01-17:
And in case you can not find the right repository:

piprom, 2021-01-21:
Hello dear oldman4U,
your last post is realy helpful for me.
Thank you. 
Have a nice day.

closed 2021-01-21 by piprom

oldman4U, 2021-01-21:
Hope they can help.

Happy printing

piprom, 2021-01-21:
Thank you my friend.
A few minutes ago I re-flash my MKS TFT24 v1.1. I check all of the possible connections around as card, but the problem was my 1GB SD card. I try with another 16GB sd card and everything is OK. What a stupid mistake. But everything in the link that you post here is very useful. Best regards and respect.