bigtreetech/BIGTREETECH-TouchScreenFirmware #1521: Bed leveling points raise Z height to 0.2

dmarnel, 2021-01-16:
### Description

Using the bed leveling points on the touch screen, the Z height at each point is 0.2

### Steps to reproduce

1. Using the touch screen, choose Movement -> Bed Leveling -> Point x
2. Return back to the main screen and notice the Z height is 0.2

**Expected behavior**
The Z height at each leveling point (corner of the bed) should be 0.0

**Actual behavior**
The Z height at each leveling point is 0.2

### Hardware Variant
TFT35 V3
### TFT Firmware Version & Main Board Firmware details
TFT: BIGTREE_TFT35_V3.0.26.x
Marlin: 2.0.7.2

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

digant73, 2021-01-16:
If you are talking about MBL menu, it is the intended result. For each point I move the nozzle to an initial Z height 0.2 just to avoid contact with the bed in case manual bed leveling was not performed so good

sarvenn, 2021-01-16:
No @digant73 . He is talking about the manual bed leveling 5 corner menu stuff. Not the manual mesh bed leveling.

@dmarnel I noticed the same before, however it has a easy solution.
In default config.ini file set the value to 0 as you may see below.

## Z Position (Manual Leveling)
# Lower z axis to this position after reaching a leveling point.
#   Format: [level_z_pos: Z<position in mm>]
#   Unit: [distance in mm]
level_z_pos:0

@digant73 I think this value should be 0 as default. It might have set to 0.2mm considering people using a thick paper to adjust their bed corners. If we set this value as 0.2mm and level the bed to almost touching to the nozzle or using a thin paper (0.1mm mostly), after homing the nozzle it will grind to bed at Z0 position in XY plane.

digant73, 2021-01-16:
manual bed leveling menu uses the configurable parameters starting with "level_" in config.ini file. As reported by @savenn, use level_z_pos to set your desired z height

dmarnel, 2021-01-17:
Thanks guys, the config level_z_pos setting worked. Was this just changed recently? I didn't have this problem until I updated the firmware. I also think having the default value as 0.2 instead of 0.0 is going to cause a lot of grief, IMO, for anyone who updates their firmware and doesn't know enough to make this change in the config file.

oldman4U, 2021-01-17:
Hi Don.

The function does exactly what is written in the config.ini, so it can not be a bug. Each function needs a default setting and of course there is always the possibility to discuss about the value itself. But believe me, changing this particular setting to 0 you get new tickets in complaining about it in no time. Why, because using a piece of paper is the recommended way to level the bed and 0.2 is the correct value for this.

Regarding config.ini. To edit it, a computer with a text editor is needed. Playing around with open source software, a computer is mandatory. So the final question is, if the users are willing to spend 5min to read, so they know what to do: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/config_instructions.md

Please help the community and close this ticket once you do not need it anymore.

Thank you and happy printing

dmarnel, 2021-01-17:
Wait, you're saying 0.2 is the correct value to use when leveling the corners with a piece of paper? I have always used a piece of paper to level my print beds, and the height of the hot end should always be the height it sits at when all axis are homed, which is 0.0. Otherwise, when the print begins, the bed will be too high because you've adjusted it with the hot end sitting at 0.2, and this can cause damage to the bed. I just had this happen which is why I am here reporting it.

oldman4U, 2021-01-17:
I do not know the code behind it but using 0.2 and a filler gauge with the same hight gives me perfect first layers. So I assume that the value defined for leveling is for leveling only and is completely ignored and therefore not relevant while printing.

dmarnel, 2021-01-17:
Well, of course it will give you a perfect bed height if you are using a filler gauge! But if you are using paper that is thinner than 0.2, like we've been talking about, you're going to gouge your bed.

closed 2021-01-17 by dmarnel

oldman4U, 2021-01-17:
In this case you can adjust the value. Whats needed is a good
documentation. Help is always welcome!