bigtreetech/BIGTREETECH-TouchScreenFirmware #1522: TFT35 E3 freezing on bed leveling menu

SebastianBraun01, 2021-01-16:
I recently upgraded my ender 5 with a skr mini e3 v2.0, a TFT35 E3 and a bltouch.
One issue i still cant fix, is the freezing of the tft when i enter the bed leveling menu. The tft crashes with a frozen screen and needs to be restarted to work again. And when it boots up i get a popup with "cant enable bed leveling".

I tried
- different bed leveling methods in the marlin firmware(manual mesh bed leveling and bilinear with the bltouch)
- different versions of the tft firmware and even compiled it myself
- different settings in the config.ini (all recomendet options in the marlin firmware are enabled)

digant73, 2021-01-16:
Try to reflash also icons

oldman4U, 2021-01-17:
Please follow the instructions here to solve the problem: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#update-tft-firmware

Please close this ticket once you do not need it anymore.

Thank you and happy printing

DrumClock, 2021-01-17:
Hi,
I don't know if it has nothing to do with it ..... in the directory
`.... \ Copy to SD Card root directory to update \ THEME_The Round Miracle Menu Material theme \ TFT35
`
the `mesh_editor.bmp` and` z_align.bmp` icons are missing.

... if I first update the bmp from the directory `THEME_Unified Menu Material theme` and then from T`HEME_The Round Miracle Menu Material theme` ..... **then UBL is accessible.**

digant73, 2021-01-17:
yes. because in the miracle theme pack  some icons were missing. there is a PR fixing that problem.
All the icons must be detected otherwise you get the issues you had.
Use the unified theme (all the icons are present) until the PR for miracle theme is merged in the main branch

SebastianBraun01, 2021-01-17:
Tried it and it works now. @DrumClock is right, mesheditor.bmp and z_align.bmp are missing not only in the round miracle theme but also in the red material theme that i used.

closed 2021-01-17 by SebastianBraun01

oldman4U, 2021-01-17:
@digant73

Would it make sense to have a spare icon for each theme which can be used as a fallback or how can we avoid such situations where icons are missing? Do you have an idea?

digant73, 2021-01-17:
yes spare icons will solve the problem