bigtreetech/BIGTREETECH-TouchScreenFirmware #1523: [BUG] Resume does not work after pausing print (SKR mini E3 2.0 TFT35 E3 V3.0)

Tetsuo456, 2021-01-17:
### Description

After pausing a print in touch mode it will do the filament purge in the last printing point instead of the top left corner and will present a resume dialog box. After that, impossible to resume by clicking on the ok button until power off.

### Steps to reproduce

1. Launch a print in touch mode
2. Pause the print

**Expected behavior**
To retract, raise the Z axis, and move to the top left corner of the bed for filament purge.
To present a dialog box with the resume question
To resume the print after clicking ok on the dialog box 

**Actual behavior**
Z axis raises, moves to the top left corner but comes back to the last printing point for filament purge
presents the resume dialog box but impossible to continue from there until power off

### Hardware Variant
TFT35 E3 V3.0
### TFT Firmware Version & Main Board Firmware details
TFT firmware BIGTREE_TFT35_V3.0.26.x (config version 20201218) - Marlin 2.0.7.2

### Additional Information
Was reported in SKR-mini-E3 as https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/488
[image]
[image]

oldman4U, 2021-01-17:
Update to firmware from yesterday and try again.

oldman4U, 2021-01-17:
You print from on board sd right?

Tetsuo456, 2021-01-17:
> Update to firmware from yesterday and try again.

I updated the firmware yesterday in fact, late at night. When was it updated?

> You print from on board sd right?

Yes, from the board's sd

oldman4U, 2021-01-17:
Do you have a tilde in the filename. Using the latest and greatest firmware you should see the full filename and not 8.3.

Tetsuo456, 2021-01-18:
> Do you have a tilde in the filename. Using the latest and greatest firmware you should see the full filename and not 8.3.

Dear @oldman4U, I checked that and updated to the latest latest, showing now the long name. However, this didn't correct the behaviour, it still does the same.
[image]

oldman4U, 2021-01-19:
Yes, I can confirm that Pausing the print using Touchscreen while printing from on board SD does not work properly. Add this to the Known Issues list.

Maybe you can change the title to include all relevant things like printing from on board SD.

Thank you

scathappens, 2021-01-19:
how do i find out when this issue is resolved, i am new around here and not sure about all this github stuff. i am experiencing the same thing? and would be keen to update my firmware when its resolved

kisslorand, 2021-01-19:
I checked also the Pause/Resume function with onboard SD printing. I did it from the TFT, from ESP3D connected to the TFT, from Octoprint (RPI->USB) and Octodash (RPI->USB).

From the TFT:
- started print:  bed, nozzle heating, print started
- hit Pause: Popup "Pause", nozzle went to front left than above where print was left off
- hit Dismiss on the popup: back to the printing screen, bottom left  button is changed to Resume
- hit resume: purge started, after the purge a popup with "Paused" message and "PurgeMore" and "Continue" buttons
- hit Continue: 6 seconds waiting and print restarted, bottom left button changed to "Pause"

I did these steps at least 5 times from TFT, ESP3D, Octoprint, Octodash. Every single time the events were the same.
Each time I monitored the serial from two sources, the TFT UART and the mainboard USB to see if the actions correspond to the messages and actions seen. They did.
   Once or maybe twice I've seen a popup with "Nozzle Parked" message when print was paused and the nozzle went to the front left. Until I pressed OK the nozzle stood there but right after the popup was gone, the nozzle went back above the printed part.
When that happened I also saw the "Nozzle Parked" on the serial monitor too.

My conclusion is that it has nothing to do with the TFT, the TFT executed everything as instructed by Marlin.
It is Marlin who is in charge of handling Pause/Resume when printing from onboard SD.

Tetsuo456, 2021-01-19:
@kisslorand could it be something to do with the ADVANCED_PAUSE_FEATURE in Marlin? Also to precise that this happens to me only in touch screen mode. When in Marlin emulation mode pause works fine.

kisslorand, 2021-01-19:
I don't know. They say it is an experimental feature.
I didn't bothered about it at all as I do not print from onboard SD and I do not change filament during print.

kisslorand, 2021-01-20:
I spent a few hours with this and found some things. [This](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1523#issuecomment-763164276) behaviour is neither the fault of TFT or Marlin.
TFT is not aware that Marlin uses ADVANCED_PAUSE and sends regular pause command and Marlin doesn't wait for user intervention after the nozzle park  if printing from onboard SD or calling ADVANCED_PARK without a parameter.
To make it work the user should tell the TFT that his Marlin is using or not ADVANCED_PAUSE, Marlin is not reporting it. I am not confident that the majority of users knows what it is so for the moment I cannot come up with a "one size fits all" solution yet.
The culprit of nozzle not staying in parked position is this `if (ENABLED(EXTENSIBLE_UI) || !sd_printing || show_lcd)` code in Marlin.
We don't have `EXTENSIBLE_UI` so `ENABLED(EXTENSIBLE_UI)` is false , we are `sd_printing` so `!sd_printing` is false and `show_lcd` is false because TFT is not using `ADVANCED_PAUSE`.
Maybe some guru around here could take the idea further, I am more than happy to share more findings or if the infos given are enough for a fix, I am happy I could inspire him.

Tetsuo456, 2021-01-22:
I understand that it is not only that Marlin is not reporting the ADVANCED_PAUSE but also that the TFT has not the UI function implemented so can only behave as it is a normal PAUSE... workaround to disable ADVANCED_PAUSE in Marlin meanwhile, for those who have no filament sensor...? What should be the behaviour of the TFT side for this function then?

guruathwal, 2021-01-26:
I tested the issue you specified but was unable to replicate it. Marlin and TFT behaved as expected and correctly preformed paused and filament change operations. 
In https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/488 the issue is caused because HOST_ACTION_COMMANDS is disabled (in Configuration_adv.h in marlin) in Marlin firmware.
It looks like in this case also you have not enabled HOST_ACTION_COMMANDS in your marlin firmware.

digant73, 2021-01-28:
In my case pausing a print simply suspend the print without parking the nozzle or providing any dialog box. Resuming the print  correctly resume the print, but if I stop the print the print is properly stopped but then the tft freezes on "Loading._." screen. The freeze is a bug on the tft fw that should be fixed. I have all the settings reported on config.ini enabled in marlin fw

digant73, 2021-01-28:
I'll do. Using recent marlin bugfix

Tetsuo456, 2021-01-29:
@guruathwal I see I have HOST_ACTION_COMMANDS disabled, so I will try to enable it to give it a try thanks.
@digant73 I have the same issue with the screen hanging at "loading" after a print stop.

Guilouz, 2021-01-30:
I have the same issue when pausing print from onboard SD.

When pressing on pause, the hotend comes to park but immediately starts again above the print (at the last position where the pause was made) and the purge is done above the print.

All features in config.ini are enable in Marlin ans I use latest 2.x bugfix

kisslorand, 2021-01-30:
The problem with this solution is that you do not know if the ADVANCED_PAUSE is enabled or not in Marlin. Marlin doesn't report it. In case ADVANCED_PAUSE is not enabled, M125 won't do anything.

oldman4U, 2021-01-30:
We have also some other dependencies. It has to be documented well.

scathappens, 2021-01-31:
so with all the issues with this, what is the way to set up the skr mini e3 v2 and the tft35 e3 v3 with a filament runout sensor so that it works in the current environment?

kisslorand, 2021-01-31:
This issue has been fixed by PR #1580.

scathappens, 2021-01-31:
ooooooo awesome....is included in new firmware or do i have to edit firmware myself?

kisslorand, 2021-01-31:
Not included yet, it needs to be merged. Wait a few days.

scathappens, 2021-01-31:
sweet thats awesome, is it just the tft firmware? i have all the features activated in my skr firmware build

kisslorand, 2021-01-31:
Yes, only the TFT firmware.

scathappens, 2021-01-31:
awesome stuff

PR #1580 (2021-02-02): Fix pausing print from onboard SD card, update French translation
Pause when printing from onboard SD card have issue when using M25 command.

Marlin related issue here : https://github.com/MarlinFirmware/Marlin/issues/18191

Solution is to use Head Park command (M125).

Before :
- Pressing pause on screen
- Print is paused and Hotend is parked but go back immediately at the last position where the pause was made without user interaction
- Purge is done above the print
- Screen tell to user to Continue

After :
- Pressing pause on screen
- Print is paused and Hotend is parked
- Screen tell to  [... cut]
PR #1588 (2021-02-02): Bugfixes, cleanup
### Requirements

BTT TFT or MKS TFT

### Description

* fixed "Loading..." screen hanging after canceling a print from onboard SD
* fixed "Loading..." screen hanging after pausing and than later canceling a print from onboard SD
* fixed nozzle moving away from parking position after "Pause" when `PARK_HEAD_ON_PAUSE` is enabled in Marlin
* fixed some memory allocation not being released after no longer needed
* fixed `request_M0` not being parsed
* fixed a hidden bug that could become active in a future development (`requestCommandInfo` is used ONLY when printing from TFT and [... cut]
closed 2021-02-02 by bigtreetech