bigtreetech/BIGTREETECH-TouchScreenFirmware #1524: [BUG] PREVENT_COLD_EXTRUSION_MINTEMP

DrumClock, 2021-01-17:
HI,
to test the extruder motors I need to set **PREVENT_COLD_EXTRUSION_MINTEMP** to **20  ° C**. 

In Marlin is a disabled feature `// # define EXTRUDE_MINTEMP 170`

In the TFT firmware, I will change the settings as follows:

[image]

[image]

1) In Marline mode everything is fine.

2) In TFT mode there is a function command from the terminal, eg G1 E100.
Via the touch screen in the `Load / Unload` menu, the temperature is still **monitored at 180 ° C**. 

[image]

Touch screen BIGTREE_TFT35_E3_V3_0

kisslorand, 2021-01-17:
Did you reload config.ini?

DrumClock, 2021-01-17:
Yes, I loaded the newly compiled FW and the config.ini file from the SD card.

kisslorand, 2021-01-17:
The value of 180 °C is loaded from the config file, it's not hard coded into the FW. Try to reset parameters.

DrumClock, 2021-01-17:
I performed **"reset default settings"** in the `parameter settings` menu ..... without change.

kisslorand, 2021-01-17:
Found it, it is hardcoded into the FW that minimal extrusion temp cannot be lower than 25 °C.

DrumClock, 2021-01-17:
in that case .... why the G1 E100 command from the terminal works even at a lower temperature ???

kisslorand, 2021-01-17:
Because it is not supervised by the TFT's FW. On terminal window the user is on his own, only Marlin remains to analyze the validity of the received gCode.

radek8, 2021-01-17:
> Found it, it is hardcoded into the FW that minimal extrusion temp cannot be lower than 25 °C. It's not a bug.

Then it doesn't make sense. You can create a PR and change the temperature to 18 ° or 20 ° to make it usable?
Thank you

radek8, 2021-01-17:
the error report is also wrong. it should say that the temperature is below 25 ° not 180 °

kisslorand, 2021-01-17:
Isn't it usable at  25 °C? What if the next guy tests his printer's extruder in his garage where is -15 °C?
What's the issue to heat the nozzle to  25 °C?

radek8, 2021-01-17:
error message says 180 :-(

DrumClock, 2021-01-17:
... even if I set the temperature to 180 and the nozzle heats up to 180 ..... there will be no filling / unloading thread .....

In Marlin mode and G-code G1 E100 is functional ... there must be an error in the FW TFT ....

kisslorand, 2021-01-17:
There's no error, everything works as programmed.

radek8, 2021-01-17:
If there is a check for a value <25, then there should be a comment in config.ini that the temperature must not be set to less than 25 °

kisslorand, 2021-01-17:
I agree, it should. There's also a max of 300 °C, just to be aware.

DrumClock, 2021-01-17:
I reset the configuration to 180 ... even if I set the temperature to 180 and the nozzle heats up to 180 ..... there will be no action .....

video: https://youtu.be/zRpzD86NI6k

rebel1, 2021-01-17:
Sorry , there is no bug

> I reset the configuration to 180 ... even if I set the temperature to 180 and the nozzle heats up to 180 ..... there will be no action .....
> 
> video: https://youtu.be/zRpzD86NI6k

Sorry, but there is no bug. 
It works as intended, in your video we can see that your current temp is 0 and 0 < 25. ;)

closed 2021-01-17 by DrumClock

kisslorand, 2021-01-17:
Just tested again "(Un)Load" menu, works as expected. I cannot reproduce any issue. (Not saying there's none)

DrumClock, 2021-01-17:
Hi, I tested again ..... the temperature T0 must be set ... If it is just heated Hotend and the temperature is not set then it does not work (probably it will not be a mistake, sorry)

I will ask:
On the TFT 50 and the old FW, this screen is:

[image]

The New TFT 35 E3 reads as follows:

[image]

I'm quite missing the "length" and "speed" icons there.

kisslorand, 2021-01-17:
In your pictures there are two different menus., one is the "Extrude" menu, the other is the "Load/Unload" menu

DrumClock, 2021-01-17:
The "Extrude" menu is on the TFT 50 display (FW from 20 Aug 2020) ...... so where can I find it on the TFT 35 E3 (FW 14.1.2021)?

kisslorand, 2021-01-17:
It is moved to the "Move" menu, the menu where you will find all 4 axis: X, Y, Z, E.

DrumClock, 2021-01-17:
### OK, thank you very much, have a nice day .....