bigtreetech/BIGTREETECH-TouchScreenFirmware #1527: SKR Pro v1.1 and TFT35 v2.0 "No printer attached"

heavylight-repo, 2021-01-17:
Hi,

I am not able to get the display to work and really needs your help! I have updated the display to the latest firmware "BIQU_TFT35_APP1_V2.0.26.x.bin" without problem. I left the display config.ini without any changes. I have also set the display baudrate to 115200.

I have made the following changes in marlin Configuration.h:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define BAUDRATE 115200

[Github help.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5826309/Github.help.zip)

What am I missing? Is there a conflict somehow with the SDsupport?

heavylight-repo, 2021-01-17:
The touchscreen works if I shift the serial cable to my TFT35 E3 V3.0...

oldman4U, 2021-01-17:
Hi.

Will you close the ticket this time once the problem is solved?

heavylight-repo, 2021-01-17:
Yes, I will. Very sorry for the last time. Had a long trip to the hospital...

oldman4U, 2021-01-17:
Hope you are doing better.

Try serial port 2 instead of serial port 1:

#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200

Let me know

heavylight-repo, 2021-01-17:
No it did not work.

These settings work for my TFT35 E3 V3.0 (in touch mode with only serial communication) on the same SKR PRO v1.1. card:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define BAUDRATE 115200

I also tried to shift the rx and tx pin if that is the difference between the TFT35 v2.0 and the TFT35 E3 V3.0...but it did not work...

Should I enable something under the LCD-settings? I know that TFT35 v2.0 does not support marlin Mode...

oldman4U, 2021-01-17:
Sorry. But then I can not help.

heavylight-repo, 2021-01-17:
OK, thank you anyway!

radek8, 2021-01-17:
Is right:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define BAUDRATE 115200
Is OK.

Connect the TFT Display to the SKR PRO
connect PC via USB
run the pronterface program on the PC and enable debug comunications
[image]

1. Send the M503 command from the PC. 
   Does the display respond - connected?
2. open a terminal on the TFT and send the command M503
   Will the PC print out a configuration report?

radek8, 2021-01-17:
Do you have another TFT display which when connected works normally?

heavylight-repo, 2021-01-18:
**With my TFT35 v2.0 display from Pronterface:**

SENT: M503
RECV: echo:  G21    ; Units in mm (mm)
RECV: echo:; Filament settings: Disabled
RECV: echo:  M200 T0 D3.00
RECV: echo:  M200 T1 D3.00
RECV: echo:  M200 T2 D3echo:  M569 S1 X Y Z
RECV: echo:  M569 S1 T0 E
RECV: echo:  M569 S1 T2 E
RECV: echo:; Tool-changing:
RECV: echo: Z0.00
RECV: ok
SENT: M105
RECV: ok T:100.00 /0.00 @:0
SENT: M105

**With my TFT35 v2.0 display from Display terminal sending M503:**

Nothing, does not respond.

**With my TFT35 E3 V3.0 (in touch mode with only serial communication) from Display terminal sending M503:**

Works, and the display respond like Pronterface.

heavylight-repo, 2021-01-18:
Maybe there are som kind of difference between the TFT35 v2.0 and the TFT35 E3 V3.0? I only run RS232 on the TFT35 E3 V3.0 and not using the EXP connectors. I did test if the RX TX was shifted but that did not work.

radek8, 2021-01-18:
it looks like a problem on the RX and TX pins of the SR232 display connector.
I'll send you a new modified FW.
You connect the RX and TX pins to the USART3 connector.
Then we will test the functionality.

heavylight-repo, 2021-01-18:
Perfekt! That would be very nice!

radek8, 2021-01-18:
I'm at work now, I'll send it tonight

radek8, 2021-01-18:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/715

There was a similar problem here and the error was in the pins of the RS232 connector.
But maybe I'm wrong ...

radek8, 2021-01-18:
If you want to test if there is no problem in the RS232 connector communication Do the following.
Upload the enclosed FW to the TFT display.
Disconnect RX and TX from the RS232 connector.
Connect the RX and TX wires to the pins of the USART3 connector.
Reconnect the RS232 connector (without RX and TX).
Turn on the printer.

[BIQU_TFT35_APP1_V2.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5831789/BIQU_TFT35_APP1_V2.0.26.x.zip)

[image]

heavylight-repo, 2021-01-18:
Thank you, I will test it tomorrow night when I get home! Respond to you then!

oldman4U, 2021-01-19:
Same hardware, problem solved by changing the bad SD card which has been used for firmware update: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/190

heavylight-repo, 2021-01-19:
WOW! USART3 works! Thank you radek8!

closed 2021-01-19 by heavylight-repo

radek8, 2021-01-19:
Caution, if you use TFT in this connection, you must not update TFT from precompiled FW (bin file).
You need to compile your own modified bin file when you swap USART3 for USART1

radek8, 2021-01-19:
> Same hardware, problem solved by changing the bad SD card which has been used for firmware update: [bigtreetech/BIGTREETECH-SKR-PRO-V1.1#190](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/190)

@oldman4U  Strange, same configuration, same problems, different solutions: - /

oldman4U, 2021-01-19:
In both cases it was not a firmware issue....

radek8, 2021-01-19:
not in this case there is a problem with the HW