bigtreetech/BIGTREETECH-TouchScreenFirmware #1528: [BUG] Filament "Load" and "PurgeMore" = reset printer

DrumClock, 2021-01-17:
if I perform the load function in the "load / Unload" menu,
after selecting the" PurgeMore" menu in the window, the printer will be restarted.

[image]

sarvenn, 2021-01-18:
@DrumClock Would you please send M600 command via terminal. Does this pop up with purge and continue options still appear?
Also would you try to print a model with M600 color/filament change command is embedded into gcode (try to use prusaslicer)? Does this pop up appear?
Unfortunately I never had this pop up appear.

DrumClock, 2021-01-18:
Hi,@sarvenn  this pop-up with the ability to clean and continue
occurs even though I use the `(No) Load` menu separately.
If I use the G-code M600 either in the file or send it via the terminal, the window will also appear.
The printer resets each time I select` "purgeMore"`.

**Is it not related to the FW Marlin requirement and not to the FW TFT?**

This is a listing of the pronterface terminal:

```
Connecting ...
Printer is now online.
>>> m600
SENDING: M600
// action: paused
// action: prompt_end
// action: prompt_begin Pause
// action: prompt_button Dismiss
// action: prompt_show
echo: busy: processing
echo: busy: processing
echo: Insert filament and send M108
// action: prompt_end
// action: prompt_begin Nozzle Parked
// action: prompt_button Continue
// action: prompt_show
echo: busy: paused for user
echo: busy: paused for user
echo: busy: processing
echo: busy: processing
// action: prompt_end
// action: prompt_begin Paused
// action: prompt_button PurgeMore
// action: prompt_button Continue
// action: prompt_show
echo: busy: paused for user
echo: busy: paused for user
echo: busy: paused for user
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
```

digant73, 2021-01-21:
@DrumClock 
did you enable all the following settings in Marlin FW:

# Options to support dialog with host:
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#   HOST_PROMPT_SUPPORT (in Configuration_adv.h)
#
# Options to support M600 with host & (Un)Load menu:
#   Options to support dialog with host (as pre requisite)
#   NOZZLE_PARK_FEATURE (in Configuration.h)
#   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)

DrumClock, 2021-01-21:
@digant73  Of course.

kisslorand, 2021-01-21:
At some point I managed to reset the printer (actually to freeze the mainboard) one single time having the popup from the first post. Unfortunately I cannot remember the steps, I couldn't reproduce it. It was while stressing the printer to see what happens between Marlin and TFT when Pause/Resume is used when printing from onboard SD.

DrumClock, 2021-01-21:
I updated FW TFT and Marlin according to the latest release ..... without change

DrumClock, 2021-01-22:
Confirmed, the error will be in FW Marlin.
I connected the TFT 35 E3 to a printer with an older FW Marlin and everything is fine there.

digant73, 2021-01-22:
Marlin fw 2.0.7.2 bugfix had a lot of bugs

DrumClock, 2021-01-22:
I use Marlin Bugfix 2.0.x  ..... How do I understand that ?

oldman4U, 2021-01-24:
Marlin like other software has bugs. The TFT firmware relies on Marlin but this repository is about TFT firmware and not Marlin.

Please help the community and close the ticket once you do not need it anymore. Thank you

DrumClock, 2021-01-24:
Hi @oldman4U 
Unfortunately, as I was excited about TFT, I'm going back to the LCD 12864. At least everything here works with Marlin.
The fact that TFT uses its own FW is great, but it's useless if it can't communicate properly with Marlin.

oldman4U, 2021-01-24:
Thank you for letting us know

Happy printing

DrumClock, 2021-01-30:
Hi @oldman4U 
**PurgeMore** still doesn't work .... tested on bugfix-2.0.x dated 2021-01-30

oldman4U, 2021-01-30:
Could you please try this firmware to see if it solves the issue: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1eb47368ea26ee08b69ca7668ac8d4b55b0409e4

DrumClock, 2021-01-31:
Hi @oldman4U  
in addition to being shuffled icons is the **purgemore** function of unchanged

oldman4U, 2021-01-31:
Update firmware config.ini and TFT folder.

DrumClock, 2021-01-31:
the update went well, only for some reason the folder was not overwritten on TFT35.CUR. I had to format the SD card.
The error is still there. PurgeMore - restarts the printer

kisslorand, 2021-01-31:
For the moment there are only 4 methods to restart the motherboard:
1. press the physical reset button
2. connect the reset pin to GND
3. send M997
4. cut the power or at least bring it under a certain voltage (be it the input voltage or the regulated 5V and/or 3.3V)

Out of these 4, the TFT is capable only to send M997 and connect reset pin to GND. Is your reset pin connected to the TFT?

oldman4U, 2021-01-31:
Please send a picture of the (Un)Load or Extrude menu after loading the firmware from the link above.

DrumClock, 2021-02-01:
HI @kisslorand and @oldman4U 
Yes, the reset pin is connected to the TFT and the board.

The Extrude menu works correctly.
(Un) Load menu after PurgeMore restarts the printer.
video: https://youtu.be/kq4-bfFMH9I

oldman4U, 2021-02-01:
@DrumClock I know that this does not work.

Please reopen this ticket. Thank you

Could you please try this firmware to see if it solves the issue: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1eb47368ea26ee08b69ca7668ac8d4b55b0409e4

Loading this firmware, the (Un)Load Menu changes to Extrude. Maybe you get a warning that two Icons are missing during FW update. You can ignore this.

Please let me know if this solves the problem with Purging or Purging more crashing the TFT.

kisslorand, 2021-02-01:
@DrumClock 
Do you have any USB device that has a LED constantly ON when connected to the USB?
If yes, please connect it to the TFT's USB and check the LED when during the same test as in your video.
If you can do it, please let me know of the results.

DrumClock, 2021-02-01:
> @DrumClock
> Do you have any USB device that has a LED constantly ON when connected to the USB?
> If yes, please connect it to the TFT's USB and check the LED during the same test as in your video.
> If you can do it, please let me know of the results.
> A USB voltage meter would be more appropriate, but a simple LED should do it.

Hi, @kisslorand 
HI, I don't have a USB device with LEDs.
I tried to connect the TFT cable from the GTR 1.0 USB board to the USB port.
The LED on the board lit up, still lit even though everything restarted.

DrumClock, 2021-02-01:
> @DrumClock I know that this does not work.
>
> Please reopen this ticket. Thank you
>
> Could you please try this firmware to see if it solves the issue: [...]

Hi @oldman4U 
HI, I compiled and uploaded the   `.pio \ build \ BIGTREE_TFT35_E3_V3_0 \ BIGTREE_TFT35_V3.0_E3.26.x.bin`  file to the TFT.
The (Un) Load Menu has not changed after pressing PurgeMore, everything still restarts.

DrumClock, 2021-02-01:
HI, I connected the TFT and the board only + 5V, GND, TX, RX.
I did not connect the RST pin. 
video: https://youtu.be/RtQ3ObcBDBk

The restart was performed anyway.

oldman4U, 2021-02-01:
As long as you have an (Un)Load menu, you are not using the firmware I sent you the link. The firmware has only a Extrude instead of the (Un)Load, and the Extrude works fine. Check Menu-Movement-Extrude and let me know if it works.

DrumClock, 2021-02-01:
Hi, @oldman4U 
I compiled FW from the link you submitted, so I don't know where the error is ????
Save the "bin" file for TFT 35 E3 V3.0 directly here and I will upload and test it.

oldman4U, 2021-02-01:
I used the precompiled firmware from the link: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1eb47368ea26ee08b69ca7668ac8d4b55b0409e4

Just copy the firmware config.ini and TFT folder to SD card and update the TFT with it.

DrumClock, 2021-02-01:
Same even if I use precompiled firmware from the link ....... PurgeMore = reset printer

oldman4U, 2021-02-01:
Please send a picture before it crashes.

DrumClock, 2021-02-01:
@oldman4U  ... picture? 
After selecting "PurgeMore" nothing happens and  restart printer.
it's in the videos above

oldman4U, 2021-02-01:
Sorry. I did not see the movie. So Extrude works (Un)Load crashes. Like on my printer.

DrumClock, 2021-02-01:
> Sorry. I did not see the movie. So Extrude works (Un)Load crashes. Like on my printer.

Yes

oldman4U, 2021-02-07:
Hi.

I believe to know what the problem is.....

Stay tuned.

oldman4U, 2021-02-08:
I have a several hours print going on on my machine, but I believe there could be a Marlin fix for the crash we see when the Purge More button is pressed. Here is the version which includes the fix for those who would like to check: 

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

DrumClock, 2021-02-08:
Hi, even the current version of "bugfix 2.0.x from 2021-02-08" does not solve anything.

PurgeMore = reset printer .... :(

oldman4U, 2021-02-08:
OK, I am still working on it.

It seems that "(Un)Load" NEVER worked on STM32 based mainboards. LPC works.

oldman4U, 2021-02-27:
The issue is likely fixed thanks to user arjanmels

https://github.com/MarlinFirmware/Marlin/pull/21210

Please update Marlin and close the ticket in case you do not need it anymore.

DrumClock, 2021-02-27:
@oldman4U 
not yet merged with bugfix 2.0.x

oldman4U, 2021-02-27:
No. But you can download his PR which includes the fix.

DrumClock, 2021-02-27:
I'd rather wait until merging with bugfix 2.0.x

arjanmels, 2021-02-27:
It was just merged. Please let me know the outcome of your testing.

oldman4U, 2021-02-28:
Hi. I did a quick test on my spare SKR PRO v1.1 with the unmerged PR and it did NOT crash anymore. ;-))) unfortunately the (Un)Load function did not work reliable but this was just a late night quick test and the test on the more popular SKR E3 DIP is completely outstanding as I have to wait for a print to finish. For me you are magician!!!

DrumClock, 2021-02-28:
Hi, Yes, it's okay, thanks.

closed 2021-02-28 by DrumClock

arjanmels, 2021-02-28:
@oldman4U and @DrumClock Thanks for testing and confirming it solves the crashes. Glad to contribute a bit to the excellent Marlin firmware.

oldman4U, 2021-02-28:
Marlin can be proud to have developers like you.

Thank you

ILGROSSOFA, 2021-05-22:
> It was just merged. Please let me know the outcome of your testing.

Hello, where can I find the modified file for solving this issue? 

If I try all display menu options "load filament", "unload filament" and "remove filament" and all of them, after asking me which type of nozzle preheat I need, the printer hangs and restart.

I have a MKS SGEN L V2 +TMC2208 V2 with TS35 display, 3dtouch, FES, PWC, dual Z.
Thank you in advance.
GROSSO

oldman4U, 2021-05-22:
This was a bug in Marlin which is fixed when using Marlin v2.0.8.1.

ILGROSSOFA, 2021-05-24:
> This was a bug in Marlin which is fixed when using Marlin v2.0.8.1.

Ok thank you! So I go to setup new Marlin firmware!

ILGROSSOFA, 2021-05-26:
> This was a bug in Marlin which is fixed when using Marlin v2.0.8.1.

I solved the problem and now it works like a charm. Thank you very much for your precious helps!

Moreover I have followed this guide to setup properly all functions related to filament change:
https://crosslink.io/2020/07/17/marlin-2-0-m600-filament-change-setup-explained/

I strongly suggested to everybody who need it.

Bye
GROSSO