bigtreetech/BIGTREETECH-TouchScreenFirmware #1529: printing stop button

3dfixer, 2021-01-17:
whilst printing from the tft menu using onboard sd, pressing stop does indeed stop print. but thats all it does.
doesn't switch off any temp or move out the way

### Steps to reproduce

press stop button

**Expected behavior**
stop print and return to non printing state  i.e. cold and head moved out of the way

**Actual behavior**
just stops movement, nothing else

### Hardware Variant
tft 35 v3.0
### TFT Firmware Version & Main Board Firmware details
marlin 2.0.7.2   tft35_v3.0.26.x jan 13 2021

### Additional Information
not a problem in marlin side, it stops print, homes x y , turns off all temps

[configs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5826555/configs.zip)

3dfixer, 2021-01-17:
OH if someone can look at my configs and tell me why i can only set extruder to 260 when i'm certain i got it set to 275.

oldman4U, 2021-01-17:
Have you tried to activate the print end function?

3dfixer, 2021-01-17:
ok searched for that. sumthing to do with password
will stopping the wife buggering about with my settings help??

oldman4U, 2021-01-17:
Password? Wife?

Not sure I understand but sounds funny😁

radek8, 2021-01-17:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1408

oldman4U, 2021-01-17:
We should rename from Cancel to Pause and maybe enable by default. There is nothing like cancel.

radek8, 2021-01-17:
It's alright.
This is not a pause of printing, but a cancellation of a running print job.
Yes, it would be good to enable it by default in config.ini.

3dfixer, 2021-01-17:
changed config.ini 
edited cancel gcode to shut everything off and move bed to my preferred position
Maybe changing name to cancel would be more understandable

3dfixer, 2021-01-17:
thanks for your assistance.
still cant figure out why i cant heat above 260, cant find a 260 anywhere

closed 2021-01-17 by 3dfixer

Tannoo, 2021-06-06:
This was closed and was not enabled by default?

```#### Default Cancel Gcode Status
#   Options: [enable: 1, disable: 0]
cancel_gcode_enabled:0```

oldman4U, 2021-06-06:
It is an option which not all users need. Why should it be enabled by default?

Tannoo, 2021-06-06:
So, it is okay to leave heaters running for the unsuspecting some?
Wouldn't this be more of a safety issue?

radek8, 2021-06-06:
A printer is not a toy for unsuspecting people.

oldman4U, 2021-06-06:
You can and should enable it in case you have no end gCode defined in your slicer.

Tannoo, 2021-06-06:
I did and I do.
But the END g-codes in the slicer are only used at the END of a print.
If the print is canceled before it is finished, the heater(s) remain ON unless `cancel_gcode_enabled`
is enabled.

I really didn't think I had to explain that.

So, nevermind.

oldman4U, 2021-06-06:
What makes you believe that you have to explain that. Now I have the feeling that I have to explain to you that this is open source and if you do not like what you get you can change it by submitting a PR. Or you do what all other users do and change the settings the way you need them.

kisslorand, 2021-06-06:
The second the cancel gCode is enabled by default there will be a debate what should that default cancel gCode be...
...and I doubt there is one that fits everyone...

So, no matter what, some degree of user customization is needed one way or the other.

Tannoo, 2021-06-06:
> What makes you believe that you have to explain that.

This:
> You can and should enable it in case you have no end gCode defined in your slicer.

Which was not the issue at all.

Anyway, the TFT's SD card is not seen by marlin when using the "TFT SD" icon. Soooo... Marlin doesn't shut off the heaters if there is an issue. That is where this gcode cancel script comes into play.
I found that I did not have the Marlin options on to enable it to see the TFT's SD card. I suspect that once I get those on, that there will be no issues as Marlin will know about printing from the SD.
I never had this issue with Octoprint. There was no cancel script there that I remember.

Thanks for all y'all's help.