bigtreetech/BIGTREETECH-TouchScreenFirmware #1532: label change and move

3dfixer, 2021-01-18:
a couple of ideas for the TFT print menu
could we change the stop label to cancel?
could we swap the stop(cancel ) icon position with the more icon.
when I'm backtracking out of the "more" submenus i often tap once too often and stop print, most annoying

oldman4U, 2021-01-18:
I hope you get a warning before the print really stops?

oldman4U, 2021-01-19:
According to Gurmeet you can change the button label in the language file.

oldman4U, 2021-01-22:
Would be great if you could answer your own ticket.

oldman4U, 2021-01-24:
Please help the community and close the ticket once you do not need it anymore. Thank you

3dfixer, 2021-01-24:
> Please help the community and close the ticket once you do not need it anymore. Thank you

why would i close it when no action has been taken???
if the powers that be don't want to rearrange the icons thats fine
and changing the language file has no effect

oldman4U, 2021-01-24:
So you think not answering the own ticket is fine?

kisslorand, 2021-01-24:
> and changing the language file has no effect

If you are using English, than go to language selection and select the second English in the list. That will use your language file, not the language encoded into the FW.

3dfixer, 2021-01-25:
> > and changing the language file has no effect
> 
> If you are using English, than go to language selection and select the second English in the list. That will use your language file, not the language encoded into the FW.

thank you, i thought just upgating language.ini did the job, didn't realize had to select it as well

3dfixer, 2021-01-25:
> So you think not answering the own ticket is fine?

really don't see what you mean by this statement.
if i could code in "C" i would not be asking for help/advice

oldman4U, 2021-01-25:
So it is ok to ignore 3 replies and not to answer for several days? Maybe I just expect too much.

closed 2021-01-28 by 3dfixer