bigtreetech/BIGTREETECH-TouchScreenFirmware #1534: Babystep function on real time printing don't save to eeprom

mpcolman, 2021-01-18:
Hi everyone, I updated with one of the latest firmware with the upgrade issue for the filament runout beeps and it work fine, but when i adjust the babystep when printing, it's dont still saved on the eprom with the modified value.

But the stored message appears but the value don't be updated.....

If I make a correction outside the babysteping option and make the modification under the menu SETTINGS-----MACHINE----PARAMETER----- probe offstet.... Z.... and save the upgrade it stored on eeprom is OK.

I'm loosing something or it's a firmware bug????
regards!

digant73, 2021-01-18:
This bug is fixed in PR #1518

mpcolman, 2021-01-18:
Bltouch method enabled...

PR #1518 (2021-01-21): Added Home Offset menu + bugfixes and cleanup
**NEW FEATURES:**
1) Added Home Offset menu. Previous ProbeOffset menu is now a generic Z Offset menu (just to not duplicate code) that will invoke Home Offset or Probe Offset API. Both the Home and Probe Offset menus are available on the Bed Leveling menu. Furthermore, Probe Offset menu is available only if probe is enabled (supported) in Marlin fw.
2) Added support to MBL Z offset (G29 S4 Zxx) in Babystep menu. Babystep menu will allow to change/save Z offset based on the following rules (matching sequence):
 *) Probe Offset: If probe is supporte [... cut]
closed 2021-01-21 by bigtreetech