bigtreetech/BIGTREETECH-TouchScreenFirmware #1536: [Q] Thermal runaway protection: Is it enabled when using the touch screen menu?

markbwalker, 2021-01-18:
Can't seem to determine if the touch screen mode has an option to enable thermal runaway protection. I think I've exhausted all the menus looking for it with no luck, and I don't see it in the config file either. Apologies if somehow I missed it.

I could use the Marlin mode since I know it's there, but I have a hitch. The SD microcard on the SKR Mini E3 V2 failed, so I use the slots on the TFT35 instead.

While I'm at it, sort of a related question: When in touch screen mode, what version of Marlin is actually being used? i.e. Does the TFT have it's own Marlin version separate from the motherboard?

Running the latest BTT TFT35 files as of 1/18/2021.

radek8, 2021-01-18:
You have Marlin in your SKR board.
TFT Marlin display does not contain. it only communicates with it via the serial interface

markbwalker, 2021-01-18:
> You have Marlin in your SKR board.
> TFT Marlin display does not contain. it only communicates with it via the serial interface

Thanks radek8,

I'm an EE w/a minor in CS, so I have a basic understanding of the hardware and the code, but I didn't know how BTT implemented things.

So, I think what you are saying is if I have thermal runaway enabled in the SKR Mini E3 V2 marlin code, that even using the touch screen, it will still be implemented when printing.  In other words, the TFT may contain some sort of Marlin code, but in the end the TFT just sends data over to marlin located on the motherboard?

Is that a correct assessment?

What confused me is when I was going thru the TFT config file, I could see some g-code commands being set (i.e. max temperatures) but struggled to determine why thermal protection wasn't part of the config.

Sorry for maybe asking some basic questions.  Can't seem to find any info when googling it.

radek8, 2021-01-18:
If your SD card does not work to update marlin, you can use the serial port to update, or if you get a Botloader that can upload Marlin via USB.
Find inspiration here:

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/346#issuecomment-761833783

radek8, 2021-01-18:
> Thanks radek8,
>
> I'm an EE w/a minor in CS, so I have a basic understanding of the hardware and the code, but I didn't know how BTT implemented things.
>
> So, I think what you are saying is if I have [...]

It's like you say

markbwalker, 2021-01-18:
> It's like you say

Thanks!

oldman4U, 2021-01-19:
Please help the community and close the ticket once you do not need it anymore. Thank you

markbwalker, 2021-01-19:
Will do, was just leaving it open for a couple of days in case anyone else wanted to comment.

oldman4U, 2021-01-19:
Perfect.

Happy printing

gordo3di, 2021-01-19:
@markbwalker we have protection for the hotend and the bed on via marlin and when it triggers the TFT will simply display a message sent from the SKR.

As for your assumption the TFT acts pretty much like an external computer or something like astrotouch so it mainly send commands to the SKR and waits for some basic responses etc. As far as I've seen there is no gcode command to enable/disable TR as a safety measure so once it's on in the firmware, its on. Originally we modified marlin to trigger a filament change (and turn off the heaters) if a runaway occurred so this way you could still salvage your print if it was some false trigger and it would display a message saying "Thermal runaway" but that's all on the marlin end (and it was marlin 1.1.9).

markbwalker, 2021-01-20:
Thanks goes out to oldman4U and gordo3di for helping to answer my questions!

closed 2021-01-20 by markbwalker

oldman4U, 2021-01-20:
And radek8!!!