bigtreetech/BIGTREETECH-TouchScreenFirmware #1538: A few issues with TFT35 e3 v3 and SKR mini E3 VE

scathappens, 2021-01-19:
SO am new to Github but seems like the place to get some answers. 

I have installed a TFT35 E3 V3 and an SKR mini e3 V2 in my printer and there are a couple of bugs i cannot work out. 

so first one is that if i am printing from the SD slot on the SKR mini then when the filament runs out i cannot resume, however that seems to be a known bug 

The next one follows on, so i have now switched to using the slot on the TFT 35, however if i am using the TFT 35 to start the print, if i cancel the print in cannot get it to run the cancel gcode to home the nozzle. 

#### Default Cancel Gcode Status
# Options: [enable: 1, disable: 0]
#
cancel_gcode_enabled:1

#### Cancel Gcode
# Run this gcode after canceling print
#
# Value range: [max 75 characters]
#
cancel_gcode: G28 X Y 

when i then cancel a print nothing happens, the print head stays where it is 

When running the print from the main board if i cancel a print it will home the head and all is well but then i loose the funtionality of pause or the runout sensor, is there a fix for either of these issues as then i will have a fully functional printer

oldman4U, 2021-01-19:
Add a M75 to your start gCode

Which firmware version are you using, send a screenshot from Info window

scathappens, 2021-01-19:
Thanks for getting back to me, really appreciate it, i have had to take a crash course in marlin to get this printer working lol, its been a rough ride 

filament runout is connected to the TFT 
SKR mini Firmware 
[image]

TFT 35 e3 V3 firmaware (i updated it to the latest version today) 
[image]

adding M75 to my start gcode will solve the problem of the cancel g code not running? or is it something else?

oldman4U, 2021-01-24:
M75 helps to recognize the sensor.

Pausing the print from on board sd does not work like mentioned before.

scathappens, 2021-01-25:
thanks for that will do some more experimenting

scathappens, 2021-02-05:
@oldman4U i think this was also to do with the other issue of the runout sensor and the onboard sd slot...that issue is apparently sorted but i am not sure if the firmware has been updated...waiting on clarification and then i will install and see

oldman4U, 2021-02-10:
Any news?

Please help the community and close the ticket once you do not need it anymore. Thank you

scathappens, 2021-02-11:
how do i close a ticket?!?!

oldman4U, 2021-02-11:
Using the web based interface you have a "close and comment" button.

scathappens, 2021-02-11:
ah ok thanks, will do, have just flashed the tft53, tested the filament runout feature and it still is not working as expected, will test the pause...am playing with my marlin config to make sure there is no issue there....so at the moment i am unsure if the problem is fixed

scathappens, 2021-02-11:
also after the filament runout has been tripped and it has notified that the heater has timed out and turned off, if i press stop i get a message saying 'busy processing, please wait'

scathappens, 2021-02-11:
Just tested the pause feature too, when i press pause it parks the print head, when i press resume it does nothing and then after 30-60 seconds show the message heater timeout, i will edit the contents of this post with the updated issues

scathappens, 2021-02-11:
i have tried to turn off the pause nozzle timout feature in marlin however when i do the build fails 

i did this by commenting out that definition
//#define PAUSE_PARK_NOZZLE_TIMEOUT

scathappens, 2021-02-11:
here is a link to my marlin build if you want to have a look through it 

https://1drv.ms/u/s!Aj2EjE__0LLQhZsJTWdvdw1O5g-V5A?e=NhBaH7

oldman4U, 2021-02-14:
Have you added M75 to the start gCode of your slicer?

scathappens, 2021-02-14:
i haven't yet...i will try this today thanks :0) BTW you are doing stirling work @oldman4U really appreciate all the help

scathappens, 2021-02-14:
i have had an interesting thing happen lately aswell, if i got to press buttons on the screen it inconsistently comes up with filament runout and then pauses the print (which i havent been able to resume from) i havent noticed a pattern to it yet so cant give you a steps to replicate but it happens from time to time when i press the 'more' buttton to then go and adjust temperature or something...i dont know if this is a known issue or not

oldman4U, 2021-02-14:
You are welcome 

Add M75 to the slicer startcode and let me know.

scathappens, 2021-02-14:
just tried it with m75 in the start gcode and it made no difference

scathappens, 2021-02-14:
am now getting that if u use the touch screen during a print it starts doing things like skipping a layer advance....i think i will re-flsh the SKR and the screen....really inexplicable where these issues are coming from but i basically cant touch it during a print at the moment or something goes wrong

scathappens, 2021-02-14:
i have reflashed the tft and formatted the print SD card, will see what happens

scathappens, 2021-02-14:
ok so if baby step it to dial in the first layer then save to eprom the next layer happens lower than it should, also if i restart the print the new layer settings are ignored

scathappens, 2021-02-14:
is there anywhere to get suport with these products as i am at my whits end, i have contacted bigtreetech direct 3 times and sent them my marlin builds as requested on each occasion etc but never heard any more from them

oldman4U, 2021-02-14:
You understand that the software is open source?

scathappens, 2021-02-14:
yeah sorry had a bad day with it all, got some help and solved todays trouble.....however the issues with the fillament runout and pause still remain.

I thought that this was a bigtreetech firmware, didn't realise all this was purely open source dev....do bigtreetech get involved at all?

oldman4U, 2021-02-15:
BTT does the hardware the community the software.

scathappens, 2021-02-16:
ah ok thats an interesting state of affairs

oldman4U, 2021-03-17:
I have been testing filament runout for several hours using the TFT firmware from yesterday and it worked great. Sensor connected to TFT - printing from TFT storage it triggers correctly and purges at the pause position. @digant73 Please mention this ticket in your PR.

Thank you

oldman4U, 2021-03-19:
I have to add, that printing from on board SD card is still not 100%, where the print resumes sometimes correctly at the first time. Sometimes it only resumes the print after the second or third time, but it does it finally. Resuming the print from a TFT storage works 100% at the first time.

Please try again and let us know

Thank you

scathappens, 2021-03-19:
Thanks for that, i will update to the latest firmware and try again over the weekend

oldman4U, 2021-03-21:
Please let us know once you tested it. What kind of sensor and how many wires?

oldman4U, 2021-03-22:
And make sure you follow the notice related to Marlin:

# Minimum Marlin firmware version: 2.0.5.4
# Distribution date: 2020-05-12
# Firmware source: https://github.com/MarlinFirmware/Marlin/releases
#
# In order the TFT firmware is able to provide all of its functionalities/features,
# ensure that the following options are enabled in Marlin firmware.
#
# General options:
#   EEPROM_SETTINGS (in Configuration.h)
#   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
#   M115_GEOMETRY_REPORT (in Configuration_adv.h)
#   M114_DETAIL (in Configuration_adv.h)
#   REPORT_FAN_CHANGE (in Configuration_adv.h)
#
# Options to support printing from onboard SD:
#   SDSUPPORT (in Configuration.h)
#   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)
#
# Options to support dialog with host:
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#   HOST_PROMPT_SUPPORT (in Configuration_adv.h)
#
# Options to support M600 with host & (Un)Load menu:
#   Options to support dialog with host (as pre requisite)
#   NOZZLE_PARK_FEATURE (in Configuration.h)
#   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
#   PARK_HEAD_ON_PAUSE (in Configuration_adv.h)
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)
#
# Options to support Babystep menu:
#   BABYSTEPPING (in Configuration_adv.h)
#
# Options to support repeatibility test to test probe accuracy (M48):
#   Z_MIN_PROBE_REPEATABILITY_TEST (in Configuration.h)
#
# Options to support aligment of multiple Z steppers using a bed probe (G34):
#   Z_STEPPER_AUTO_ALIGN (in Configuration_adv.h)

Please let me know

Thank you

oldman4U, 2021-04-08:
@Msq001 I believe all Pause and filament runout issues are solved now but this user does not reply anymore.

Thank you

closed 2021-04-08 by bigtreetech