bigtreetech/BIGTREETECH-TouchScreenFirmware #1539: [BUG] update firmware cause issue with input when screen was rotated.

BlackDragon2020, 2021-01-19:
I am just now getting back at it. Still a bit too busy to be of any help, but think may have found a bug. I will test for sure with in a few days. I just wanted to get it down incase I forget.
### Description

I had my screen rotated last time it was all up and running. I just updated to the newest version of BIGTREE_TFT35_V3.0_E3.26.x.bin from the pre made folder. the screen did rotate back to normal but it seems to remember the autocalibration from being flipped. top left button was giving me the bottom right button.  

### Steps to reproduce
I would think  rotate screen 
update firmware

**Expected behavior**
screen go back to stock

**Actual behavior**
screen goes back to stock, but the input for the touch is still filliped. 

sorry i dont have more info. I will get around to testing this more latter.

oldman4U, 2021-01-19:
@rixnco

Could this be related to your code?

oldman4U, 2021-01-19:
BlackDragon2020. You should change your nickname to 2021 maybe and try this code to see if it solves your issue:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1ceeea24ebe488e791abcf9cf7c35b13366ce9de

This is the current master minus the code from rixnco

rixnco, 2021-01-19:
Indeed, this is related to the bug I corrected. I've not corrected the case where you rotate the screen then update to a new firmware where the PARA_SIGN changed.
I'll make a PR to correct this other bug

BlackDragon2020, 2021-01-19:
I used the premade files from 4 days ago, as I need to redo my vs code. HDD/sdds became full.

seems i cant get it to save my config after updating the points, or rotating then rotating back. every power off i have to redo the points.

I will get it set up soon and test the code above and the current build.

commit 4cd59f1c (2021-01-21): FIX #1539 (#1541)
closed 2021-01-21 by bigtreetech