bigtreetech/BIGTREETECH-TouchScreenFirmware #1540: [BUG] BIGTREE TFT35 V3.0 backlight slowly turns off during firmware update

dreyco676, 2021-01-20:
### Description

### Steps to reproduce

1. Put BIGTREE_TFT35_V3.0.26.x.bin, config.ini and TFT35 directory (any) on either 128MB SD or 8GB SD card
2. Power up with SD card in place. Starts loading
3. As it starts loading words the screen slowly dims until it is completely off at about 40%.
4. Waiting 5min the screen is still off and firmware is not loaded on restart

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details 

### Additional Information
* Side note the D2 LED on the PCB is red and the D1 is green.

radek8, 2021-01-20:
Sd card format on FAT32, or use another card.

dreyco676, 2021-01-20:
They are already FAT32 and I've used two cards.

dreyco676, 2021-01-20:
Tried a third card. Backlight completely turns off by 46%

dreyco676, 2021-01-20:
See video

gordo3di, 2021-01-20:
Very weird. I have 5 SKR turbos with the TFT35 v3's and Marlin 2.0.5 and I just updated all the firmware yesterday (January 19th 2020) and have never seen this issue. I do compile my own firmware and I don't use the precompiled files. I use class 4 4gb cards formatted as FAT32. Have you tried powering the screen without the SKR and updating?

dreyco676, 2021-01-20:
[image]

dreyco676, 2021-01-20:
I haven't tried that. How do I power it without the main board?

dreyco676, 2021-01-20:
[image]
[image]
[image]

dreyco676, 2021-01-20:
I tried starting it up without the EXP1 & EXP2 plugged in, it gets a bit further, 49% before the screen turns off.

gordo3di, 2021-01-20:
I believe the board has a 2pin 5v connector that's just for powering the lcd. Check the pinout from biqu.

dreyco676, 2021-01-20:
Is it possible to update firmware via USB instead of SD?

oldman4U, 2021-01-20:
Sd card only.

Cut a USB cable and connect 5v and gnd to the corresponding pins of the RS232 connector. Make a connector or insulate the cables well to not shorten them.

dreyco676, 2021-01-20:
I think the screen backlight is just broken. I got it to boot up without the update and it does the same fade out.

oldman4U, 2021-01-20:
Try what I wrote and you will know.

dreyco676, 2021-01-20:
I'll give it a shot.

dreyco676, 2021-01-20:
Ok, so turns out you can power it over USB. I was able to successfully update the firmware. Looks like now I have to figure out why the touch screen is unresponsive but that's another issue.

dreyco676, 2021-01-20:
Screen still does not stay back lit when connected to SKR

dreyco676, 2021-01-20:
Seems like maybe I have a bad cable?

oldman4U, 2021-01-20:
Or a bad board.

What means non responsive? Reset the TFT.

dreyco676, 2021-01-20:
I've tried resetting (with the button). Unresponsive means it doesn't react to touch, I've even tried detaching it from my bezel to ensure its not being pressed on. The dial works however and i can get into Marlin mode.

Codel1417, 2021-01-20:
add a file to the sd card called reset.txt and reboot to fix touch screen
not responding

oldman4U, 2021-01-20:
Read the ReadMe from front page.

dreyco676, 2021-01-20:
Resetting with reset.txt forced a reflash of firmware. Still same issues. Going to chaulk it up to a bad board.

oldman4U, 2021-01-20:
You should be able to press buttons on touchscreen as soon as it gets
power. No mainboard has to be connected for this.

dreyco676, 2021-01-20:
Yea no dice. Trying that either.

dreyco676, 2021-01-21:
Ordered a new TFT, if that has no issues we'll say it was just a bad board and I'll close issue.

oldman4U, 2021-01-21:
Which files have you had on sd card during firmware update?

Greendog2020, 2021-01-21:
In my case, when I update the firmware, as I have "Auto power" enabled and the "bigtreetech relay module" connected to the screen, I have to reset the relay every 15 seconds or so until the firmware is installed, otherwise it would happen to me like you.

dreyco676, 2021-01-21:
@oldman4U BIGTREE_TFT35_V3.0.26.x.bin, config.ini and TFT35 directory

@Greendog2020 where is "auto power" configured? I'm just using the precompiled firmware.

oldman4U, 2021-01-21:
And everything updated without an error?

TFT does not dimm using USB power?

dreyco676, 2021-01-21:
Correct everything updated without error. TFT does not dimm with USB power
and screen is much brighter.

oldman4U, 2021-01-21:
Remaining issue is the inability to press buttons?

Anything else?

dreyco676, 2021-01-21:
Yes touchscreen is unresponsive no matter what and without external USB
power the display slowly dimms until it is completely black.

oldman4U, 2021-01-21:
Sorry to ask again.

With ONLY USB power, the display still dimms?

dreyco676, 2021-01-21:
With only USB power, display does not dimm and touchscreen still does not
respond.

oldman4U, 2021-01-21:
After reset you got a calibration screen?

dreyco676, 2021-01-21:
After I reset with the reset.txt I got "illegal flash app" that wouldnt go
away until I reflashed the firmware.

oldman4U, 2021-01-21:
You should read the ReadMe.

After flashing the  firmware, remove the sd card and restart the TFT.

Insert sd card with reset and see.

dreyco676, 2021-01-21:
I've read the readme and followed the steps you outlined.. When I put only
the reset.txt on the SD card after restarting that is when I got a illegal
flash app message showing up.

oldman4U, 2021-01-21:
You get the message in case a firmware.cur file is on the sd card at the
first start after the Update

dreyco676, 2021-01-22:
Reformatted and tried several cards. Reset finally worked. We are all good now.

closed 2021-01-22 by dreyco676

oldman4U, 2021-01-22:
Almost expected this 👍🏻

Happy printing