bigtreetech/BIGTREETECH-TouchScreenFirmware #1542: TFT35 E3 No Media Marlin Mode

Doyle4, 2021-01-20:
### Description

Screen was displaying SD Init Fail, updated to latest build today (says from 4 days ago) now displays No Media

### Steps to reproduce

Insert SD/USB Stick, Marlin Mode, Says No media

**Expected behavior**
To see files

**Actual behavior**
Screen displays No Media

### Hardware Variant
TFT3.5 E3 V2 I think it is? Dial at bottom right
### TFT Firmware Version & Main Board Firmware details

TFT
9b762cf 4 days ago

Marlin - Latest Bugfix 2.0.7.2

### Additional Information
[Configs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5841894/Configs.zip)

oldman4U, 2021-01-21:
Using your config, you can print from TFT SD card and nothing else. Is this what you want? Please check and let us know.

Doyle4, 2021-01-21:
Hello @oldman4U Thanks again for advice and help.

I print from either SD or USB from TFT currently which is what Id like.

The onboard SD port on motherboard I only use for updating the board :)

Doyle4, 2021-01-21:
Do I have it to Print from SD mainboard?

oldman4U, 2021-01-21:
I BELIEVE printing from TFT in Marlin mode is only available in case you have EXP1 and EXP2 connected, but I can for the person who really knows.

@radek8. Is it possible to print from another media than on board SD in Marlin mode and does this depend on a certain connection?

oldman4U, 2021-01-21:
doyle4, which mainboard do you have?

Doyle4, 2021-01-21:
@oldman4U I have the MKS SGEN L V2.

I have a microsd extension im happy to use if it has to be from board.

radek8, 2021-01-21:
he has BOARD_MKS_SGEN_L_V2

radek8, 2021-01-21:
[image]

radek8, 2021-01-21:
Do you have an EXP2 connector connected?

Doyle4, 2021-01-21:
Yeah, I have both connected.

radek8, 2021-01-21:
The marlin is a parameter
#define SDCARD_CONNECTION
which can have 3 values.
LCD - in both Marlin mode and touch mode, only the card on the TFT display is available, which is connected via the EXP2 connector. The SD card on the MKS is only available for Marlin updates, otherwise it is inaccessible.

ONBOARD - In Marlin mode, only the SD card on the MKS board is available.
Both TFT and MKS cards are available in touch mode. You can use both cards.

CUSTOM_CABLE - An SD card is available which is user-defined on other ports in the pin definition file.

Is it that understandable?

radek8, 2021-01-21:
I prefer the ONBOARD parameter
I can then use both SD card slots.
Unfortunately, Marlin mode only supports one SD card.

Doyle4, 2021-01-21:
If i was to change it from TFT to onboard, could I still use USB port? And if changed to onboard, can I still update via onboard sd? I know it doesn't mount when changed to onboard, thats the part that puts me off.

radek8, 2021-01-21:
USB is still available. USB does not affect this setting.

Marlin is always updated to MKS from the onboard SD card.
Upgrading from a reader to a TFT would only be possible when loading an alternative botloader.
There is an alternative botloader for SKR, I don't know for MKS, I didn't need it.

Doyle4, 2021-01-21:
So I never use the TFT sd, always a sd to usb stick as one on my computer has broke.

Would like to be able to print from Marlin mode, Im going to change it to Onboard and use a MicroSD extention adapter I have, my sd port is covered and little further back after upgrade.

radek8, 2021-01-21:
It's your choice. If it doesn't suit you, you can always put it back

Doyle4, 2021-01-21:
Thanks @radek8 so I need to change it from LCD to ONBOARD for access in Marlin mode, few options I like in Marlin mode so Im happy to change.

Spotted also, USB did not work in Marlin mode either, but that doesnt bother me, as long as SD works im happy.

radek8, 2021-01-23:
USB to TFT will never be accessible in Marlin mode

oldman4U, 2021-01-23:
Please help the community and close the ticket once you do not need it anymore. Thank you

Doyle4, 2021-01-23:
Onboard solved it thanks guys!

Sorry for late close, not been around much sorry.

closed 2021-01-23 by Doyle4

reallost, 2022-12-31:
I have and skr1.4 marlin 2.0.5 with tft35. Sd and usb work fine in marlin mode. 
i also have a skr 3ez marlin 2.1 bugfix and tft35. Both usb and sd dont work in marlin mode. Have swapped screens and nothing changes.  Both screen recognize in touch mode.  

> USB to TFT will never be accessible in Marlin mode
 So this isnt really a true statement