bigtreetech/BIGTREETECH-TouchScreenFirmware #1543: [FR] when starting print set both bed and hot temp same time.

BlackDragon2020, 2021-01-20:
**Is your feature request related to a problem? Please describe.**
If I do not hit preheat first before starting a print, I have to wait for the the bed or the hotend to heat up before it will start heating up the other. I do not recall what one it starts first.
**Describe the solution you'd like**
I have not looked at the code yet, but I we can read both the set temp for hotbed and  hotend then send both commands. It seems it is sending it like a print point gcode. aka do one action wait for it to complete before doing the next.

we may have to add a start up function be that reads the code for the temps and sets them before it goes in to normal print mode if the code is complex.

Just my thoughts I will try to work on it at some point i hope.

kisslorand, 2021-01-20:
> If I do not hit preheat first before starting a print, I have to wait for the the bed or the hotend to heat up before it will start heating up the other. I do not recall what one it starts first.

It is so because of your gCode. You cannot blame TFT that it does what the gCode says.

BlackDragon2020, 2021-01-20:
> > If I do not hit preheat first before starting a print, I have to wait for the the bed or the hotend to heat up before it will start heating up the other. I do not recall what one it starts first.
> 
> It is so because of your gCode. You cannot blame TFT that it does what the gCode says.

Thank you for the reply. I dont recall how the old screen worked. I will install it and see what it does. It may have compensated, maybe not. I will also look into my slicer.

If tft display functions like the original, I will consider this closed, but if it does not I would consider this a function the display is missing. If you get what I mean .

kisslorand, 2021-01-20:
Just check your gCode for M109 and M190.

oldman4U, 2021-01-21:
BlackDragon2020

Whatever your old display does, the TFT works correctly by executing gCode which comes from the slicer, like kisslorand says. This is neither a bug nor a feature request, because it does what it has to do.

Please close this ticket, thank you

radek8, 2021-01-21:
If you do not want to wait for the hotend to heat up before the bed heats up, try enabling the start code and setting the hot end preheating to 180 °, for example.
Then heating to the final temperature would take less.

closed 2021-01-21 by BlackDragon2020