bigtreetech/BIGTREETECH-TouchScreenFirmware #1544: [FR] make the hot end, bed, fan, speed, etc info disyplay boxes in print mode in to buttons.

BlackDragon2020, 2021-01-20:
if you click the hot end or fan area not in print it will take you to the menu of that item to change the setting.

during a print touching them does nothing you have to go to the menu to get to them.

It would be nice if the gui worked the same in print and out of print.

thanks

I again I will try to lookin to it, but for now I am just getting the idea out there.

kisslorand, 2021-01-20:
It would be nice to do a  search before you make a FR.
Check FR #1506.

BlackDragon2020, 2021-01-20:
> It would be nice to do a search before you make a FR.
> Check FR #1506.

noted, i jumped the gun, sorry.

closed 2021-01-20 by BlackDragon2020

oldman4U, 2021-06-15:
@BlackDragon2020 Maybe you want to try kisslorands PR which brings the requested feature.