bigtreetech/BIGTREETECH-TouchScreenFirmware #1545: [BUG] TFT 70 seems to always think no 3d printer is connected

Slenderman00, 2021-01-21:
### Description
It says "no printer attatched" but behaves as it should when connected to a printer.

### Steps to reproduce

1. Connect to a rambo 1.4 board running marlin

**Expected behavior**
I expect the message to dissapear once it has connected to the printer.

**Actual behavior**
It successfully connects to the printer but the message remains

### Hardware Variant
TFT70
### TFT Firmware Version & Main Board Firmware details
tft version 3.0.26
marlin version 2.0.7.2

oldman4U, 2021-01-21:
Hello Joar.

And you believe the problem is TFT firmware related?

Slenderman00, 2021-01-21:
> Hello Joar.
> 
> And you believe the problem is TFT firmware related?

Yeah, everything works except for the "printer not attached" message

kisslorand, 2021-01-21:
Try this:
Keep holding an icon (except the first two from the upper left corner) until you get into the Marlin/Touch mode selection menu and select Touch. If you do this, do you still have the "No printer attached" message?

Slenderman00, 2021-01-21:
> Try this:
> Keep holding an icon (except the first two from the upper left corner) until you get into the Marlin/Touch mode selection menu and select Touch. If you do this, do you still have the "No printer attached" message?

It still says no printer attached

radek8, 2021-01-22:
You have commands enabled in Marlin
M503, M115, M114?
That will be your problem

Slenderman00, 2021-01-24:
It was caused by line 315 in in parseACK.c
"if (!(ack_seen("@") && ack_seen("T:")) && !ack_seen("T0:")) goto parse_end;"
now im recieving the error "Unknow command M221 D0" constantly

Slenderman00, 2021-01-24:
It seems that the issue is related to my board beeing configured as a 3D printer.
for anyone else looking to do the same the issues can be solved by commenting out line 315 in parseACK.c
and removing all instances of storeCMD("M221")

closed 2021-01-24 by Slenderman00