bigtreetech/BIGTREETECH-TouchScreenFirmware #1547: [Q] (Your Question?)

Patrick-81, 2021-01-21:
**Your question or details you need. Please describe.**
While printing, the speed of fan is not displayed. It is only displayed if I change the speed manually. Is it normal or have I something to set in the firmware to get it working ?

**specify the device variant related to your question.**
TFT 3.5 E3 V3.0

**Additional context**
Add any other context or screenshots about the question here.
[image]

oldman4U, 2021-01-21:
Hi

Update to master firmware which is less than 24h old

radek8, 2021-01-22:
Read the comments in Config.ini
do you have it set in Marlin?

In order the TFT firmware is able to provide all of its functionalities/features,
ensure that the following options are enabled in Marlin firmware.

General options:
  M115_GEOMETRY_REPORT (in Configuration_adv.h)
  M114_DETAIL (in Configuration_adv.h)
  # REPORT_FAN_CHANGE (in Configuration_adv.h)
  EMERGENCY_PARSER (in Configuration_adv.h)
  SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
  HOST_ACTION_COMMANDS (in Configuration_adv.h)

Options to support printing from onboard SD:
  SDSUPPORT (in Configuration.h)
  AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
  AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
  LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
  SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

Options to support (Un)Load menu:
   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)

Patrick-81, 2021-01-22:
Hello, I have updated the TFT 3.5 firmware with same result. I will configure the card firmware as suggested by radek8 and make a test. Thank to both of you.

radek8, 2021-01-22:
[image]

Patrick-81, 2021-01-22:
this is what i have
/**
 * EEPROM
 *
 * Persistent storage to preserve configurable settings across reboots.
 *
 *   M500 - Store settings to EEPROM.
 *   M501 - Read settings from EEPROM. (i.e., Throw away unsaved changes)
 *   M502 - Revert settings to "factory" defaults. (Follow with M500 to init the EEPROM.)
 */
#define EEPROM_SETTINGS       // Persistent storage with M500 and M501
//#define DISABLE_M503        // Saves ~2700 bytes of PROGMEM. Disable for release!
#define EEPROM_CHITCHAT       // Give feedback on EEPROM commands. Disable to save PROGMEM.
#define EEPROM_BOOT_SILENT    // Keep M503 quiet and only give errors during first load
#if ENABLED(EEPROM_SETTINGS)
  #define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.
#endif

radek8, 2021-01-23:
Are no values stored, or just a jerk value?

Patrick-81, 2021-01-23:
Looks like only the jerk values are concerned.
Another issue is the extruder calibration which looks quite impossible from the TFT and must be done from Marlin screen. Asking 100mm extrusion from TFT never gave me the good value. A step by step of 10mm looks to work finer after calibration under Marlin. Will have to check if there is an issue open on it, and will try to reproduce it later.

radek8, 2021-01-23:
In this case, you do not have CLASSL_JERK enabled in Marlin and you use JUNCTION_DEVIATION.
Only one of them can be active.

Patrick-81, 2021-01-23:
yes JUNCTION_DEVIATION is active. Which option is it the best ?
Seems that JUNCTION_DEVIATION doesn't allow any customization.

radek8, 2021-01-23:
It depends on personal preferences.
Use it to get better results

Patrick-81, 2021-01-24:
Thank you for your help Radek8.

closed 2021-01-24 by Patrick-81