bigtreetech/BIGTREETECH-TouchScreenFirmware #1549: TFT35 E3 V3 connected to SKR mini E3 v.2 on an ender 3 pro

t0m541, 2021-01-22:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

t0m541, 2021-01-22:
Tried updating the screen firmware, the firmware, config and language updated ok, but now I have the message on the screen saying that the icon and fonts are outdated, even though I copied across the latest icon and font file from the round theme, which had the same update date as the rest of the files.
I'm new to this, can someone give me some direction on what I'm doing wrong.
Even retro date it back to it's shipped software.
The build date was 26/12/2020

oldman4U, 2021-01-22:
Read the ReadMe at the main page and use the master firmware which is maximum 2 days old.

t0m541, 2021-01-22:
I used the bigtree-tft V3 26... Firmware to flash to the screen, that flashed ok along with the language and config.
All files were 2 days old which is why I can't understand why the screen is saying the fonts and icons are outdated.
I decided to try the update as the original screen software seemed very glitchy it kept freezing and the icons touch points were way off even after doing the touch calibration

t0m541, 2021-01-22:
Could it be that the icon and font files were made slightly before the firmware so the screen sees it out of date, even though it could be by hours ?

oldman4U, 2021-01-22:
I updated 2 screens successfully today. Try another sd card.

t0m541, 2021-01-22:
I'm using the SD card that BTT supplied.
I will try another card.
The FW, language and config went without a hitch.
I've also fitted a bl touch, which is possibly another glitch, but I'll comment that on the Mini Skr hub

t0m541, 2021-01-23:
Hi, thanks for the input and help.
This morning I tried another sd card and also used my win 10 machine, (my daily driver is Linux) to download the zip file and transfer the files across to the card.
It seems to have fixed the update issue.
It's a learning curve

oldman4U, 2021-01-23:
Sounds good 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-01-23 by t0m541