bigtreetech/BIGTREETECH-TouchScreenFirmware #1551: [BUG] TFT70 will not operate in touch screen mode

Tuckerkj, 2021-01-23:
### Description

The TFT70 connected to a SKR 1.4 Turbo will operate normally when in LCD 12864 mode and will beep when in touch screen mode, however, no response from actions and screen reports "no printer attached".

### Steps to reproduce

Connect TFT70 to SKR 1.4 Turbo

**Expected behavior**
TFT70 to operate either as a touch screen input or as a rotary knob input.

**Actual behavior**
Printer is fully functional under the LCD 12864 mode.  Touch screen will make beeping noises when pressed without response.

### Hardware Variant
TFT70 v3.0
### TFT Firmware Version & Main Board Firmware details
TFT 3.0.26.x from BEFORE the 20 JAN 21 update and Marlin 2.0.7.2

### Additional Information
I'm using a BTT TFT70v3.0 on a BTT SKR 1.4 Turbo and both BTT devices have the latest firmware (2.0.7.2 for the SKR and 3.0.26.x for the TFT).  The printer will function only using the Marlin LCD 12864 interface and the touch screen mode will not work.  When in the touch screen mode, the touch screen will beep in response to pressing a button, without corresponding action from the printer.  Also, the screen reports "No printer attached".  Holding the rotary button results in a beep followed by the selection screen between the Marlin mode and touch screen mode.  I have ensured that the RS232 ribbon cable and both EXPx cables have continuity through all ends, are connected to the board properly, and are hot-glued into position.

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5859229/Configuration.zip)

kisslorand, 2021-01-23:
Did you try with the "reset.txt" on the SD card?

oldman4U, 2021-01-23:
Try to disconnect the RS232 cable from the TFT to see if the beed stays.

Update was successful for ALL elements?

Tuckerkj, 2021-01-25:
Gents, my apologies for responding sooner but SUCCESS!!  Both devices would update firmware without error, however, after disconnecting the RS232, the TFT would fail on reset.txt.  Meaning I would have to attempt the calibration 4-5 times before it finally took.  Reconnect the RS and the TFT actually connected.  Two to three second delay before it would connect but connect it did.

New question.  Is it normal for there to be this 2-3 second delay for the TFT to "connect" to the printer when the printer is turned on?

Cheers!!

oldman4U, 2021-01-25:
Sounds good and all you describe is normal.

Happy printing 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-01-25 by Tuckerkj