bigtreetech/BIGTREETECH-TouchScreenFirmware #1554: [Q] Touch screen calibration fail

luwi66, 2021-01-23:
I try to install the TFT35-E3 V3.0 to my printer.
Firmware update (V26.1) ok. Touch screen calibration doesn't work. I can touch three red dots, but the fourth red dot (lower left corner) doesn't show. So I always get "Adjustment failed. "

[image]

sarvenn, 2021-01-23:
Because there isn't any red dot on the lower left corner. You need to press on the middle black dot after tapping on to lower right red dot.

oldman4U, 2021-01-24:
Please help the community and close the ticket once you do not need it anymore. Thank you

luwi66, 2021-01-24:
That’s exactly what I did 20 times. But it would be nice if you could change the text to say „press black dot“ after the third red dot.OK, I got it working now. After many many tries with tipping to various spots suddenly SUCCESS. I can’t say what exactly caused the positive result. And I will not try to test calibration now it’s ok.I will close the ticket.Thanks  Gesendet von Mail für Windows 10 Von: sarvennGesendet: Samstag, 23. Januar 2021 17:31An: bigtreetech/BIGTREETECH-TouchScreenFirmwareCc: luwi66; AuthorBetreff: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] [Q] Touch screen calibration fail (#1554) Because there isn't any red dot on the lower left corner. You need to press on the middle black dot after tapping on to lower right red dot.

closed 2021-01-24 by luwi66