bigtreetech/BIGTREETECH-TouchScreenFirmware #1555: [FR] Full Klipper Support

CraXyOW3, 2021-01-23:
**Is your feature request related to a problem? Please describe.**
TFT Screen will not work with Klipper out of the box, simulated can be fixed.

**Describe the solution you'd like**
To implement serial pass-through or serial listener, so Klipper remains in control but can take commands.
In short, make a proper emulator and make the touchscreen work with Klipper.

**Describe alternatives you've considered**
Even a generic mode would be ok, so more than Klipper can utilize the screens.
An intermediary config file for translating commands, i.e screen wants G29, but when pressed will send "BED_MESH_CALIBRATE" instead, so config file could be G29=BED_MESH_CALIBRATE instead of standard G29=G29.

**Additional context**
There has been discussions about getting the emulator to work an @teeminus has gotten it to work decently here https://github.com/teeminus/BIGTREETECH-TouchScreenFirmware
Also there is/was progress being made here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290
Then we have one line that i was curious about here -> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/search?q=klipper+repo%3Abigtreetech%2FBIGTREETECH-TouchScreenFirmware
So there might have been a tought of adding support for Klipper?

Might add that Klipper is growing and is just such a perfect firmware, i do not ever believe that it will be less used. Definitively in the custom printer market.

I bought one of these tft screens well knowing it might not work flawlessly OOTB.
But surely you could make the GUI be able to pass commands.

Big shoutout for @teeminus for making the LCD usable for me!

oldman4U, 2021-01-23:
Please see #290 and also the pinned FR ticket.

CraXyOW3, 2021-01-24:
Yes, I did read the pinned ticket, which links to #290 as I also did in my FR.
My purpose of this ticket is not only basic support for the emulator, but also support for the touchscreen/GUI.
Sorry if I didn't make myself clearer.

oldman4U, 2021-01-24:
The basic support did not happen in more than one year. Hmmmm. Lets see.

closed 2021-04-01 by stale[bot]

ihrapsa, 2021-08-16:
Hey, not sure if you're aware of [Desuuuu'](https://github.com/Desuuuu)s Klipper [fork](https://github.com/Desuuuu/klipper) that allows a type of dwin TFT serial display to work with Klipper directly through mainboard, well now he's cooking something else 😄. He's working on a sort of serial pass-through implementation that will allow dynamically control of TFTs or basically any MCU directly from the klippy host through the mainboard's secondary serial interface. He said that this will work in a similar way Klipper already works with I²C and SPI.

If anyone's interested in adding touchscreen support for Klipper, check his work on the [runtime-uart](https://github.com/Desuuuu/klipper/commits/runtime-uart) branch. Maybe someone could use that to finally port the btt touchscreen firmware in Klipper