bigtreetech/BIGTREETECH-TouchScreenFirmware #1559: [Q] is Smart filament sensor working connected to TFT?

killrob, 2021-01-24:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

oldman4U, 2021-01-25:
Why not asking #GenaUser who opened a few smart filament sensor related tickets recently!?

killrob, 2021-01-25:
> 
> 
> Why not asking #GenaUser who opened a few smart filament sensor related tickets recently!?

cause i didn't know that #GenaUser do that

oldman4U, 2021-01-25:
But now you found it, great!

killrob, 2021-01-25:
> 
> 
> But now you found it, great!

yes but.... maybe i will dig a little on the code ;)

digant73, 2021-03-30:
If you solved the problem, please close your bug report

closed 2021-03-30 by killrob