bigtreetech/BIGTREETECH-TouchScreenFirmware #1564: Persistent pop up, infinite extrusion, stepper motors locked when print is canceled

KiingYS, 2021-01-25:
My setup:
SKR mini E3 V2.0 
Marlin 2.0.7.2 (Dec 3rd 2020)
TFT70 V3.0 
TFT70 V3.0.26.X (Jan 5th 2021)

1. Persistent pop up
After automatically loading the filament, a pop up says "Paused for user, continue?"
If I click ok I can't dismiss the notification.

### Steps to reproduce

1. Menu
2. Movement
3. (Un)load
4. load

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

2. Infinite extrusion
### Steps to reproduce
Menu
extrude
load
After user clicks "load", it starts loading but never stops.

3. Stepper motors locked after print is canceled.
This one is probably not a big deal, but worth looking into. When the print is paused, the motors have to be locked to prevent offsets. But when canceled, I'm not sure if it's necessary to lock the steppers.

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly dem
[1.3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5869353/1.3.zip)

onstrate the issue.

oldman4U, 2021-01-25:
You print from on board sd?

sarvenn, 2021-01-26:
I have the same issue, persistant pop up. I never asks for purge more and continue. I print from TFT SD.
[image]

Andy-ABTec, 2021-01-26:
Looks familiar +1

3dfixer, 2021-01-26:
cant remember how i got rid of this problem, but when its happening if you turn knob a couple clicks then push button, it will carry on like normal. unless you turned knob in wrong direction, in witch case it will purge some more

digant73, 2021-01-26:
verify on Marlin fw you have all the following setting enabled:

# Options to support dialog with host:
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#   HOST_PROMPT_SUPPORT (in Configuration_adv.h)
#
# Options to support M600 with host & (Un)Load menu:
#   Options to support dialog with host (as pre requisite)
#   NOZZLE_PARK_FEATURE (in Configuration.h)
#   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)

KiingYS, 2021-01-26:
> You print from on board sd?

Yes, I do

sarvenn, 2021-01-27:
> verify on Marlin fw you have all the following setting enabled:
> 
> Options to support dialog with host:
> EMERGENCY_PARSER (in Configuration_adv.h)
> SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
> HOST_ACTION_COMMANDS (in Configuration_adv.h)
> HOST_PROMPT_SUPPORT (in Configuration_adv.h)
> 
> Options to support M600 with host & (Un)Load menu:
> NOZZLE_PARK_FEATURE (in Configuration.h)
> ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
> FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)
> 
> it could be also a problem on the knob. In my case the knob button remained clicked

I use precompiled firware in SKR mini e3 repository. (firmware.bin) So I don't know it's configurations. But I noted this, this weekend I will learn to compile my own Marlin and check these items.

Andy-ABTec, 2021-01-27:
I was missing HOST_PROMPT_SUPPORT - Currently recompiling, we'll see if it helps...

kisslorand, 2021-01-27:
So? Did it help?

Andy-ABTec, 2021-01-28:
Cant be 100% sure but it seems to have done...

oldman4U, 2021-01-28:
sarvenn According to the picture you have attached, your TFT shows the PRE print screen and not the print screen. How have you started the print?

Andy-ABTec, 2021-01-28:
No, I was wrong, the load appears to complete ok, then get a popup asking purge more/continue? Select purge more and my TFT35 E3 V3 reboots.
I also pulled a fresh copy of the code from BTT and compiled it. Compiled ok, but when I try to install it only the bin gets updated, everything else is left as is.

sarvenn, 2021-01-28:
> sarvenn According to the picture you have attached, your TFT shows the PRE print screen and not the print screen. How have you started the print?

In the picture I sent a M600 gcode from my custom gcode menu. I also tried to do a (un)load during printing and a color change command integrated gcode file (sliced with PrusaSlicer), they act all the same. I never could get a pop up with purge more and continue options. But If I switch to marlin mode during the pop up above, purge more and continue options appear in marlin mode, then I select continue and switch to TFT mode.
I've shared them at #1210

oldman4U, 2021-01-28:
I can reproduce the crash.Avoid it by not purging at the moment. 

Once you reach the point where you can not continue by pressing OK - move the encoder one step clockwise and press the encoder once. Let me know

Andy-ABTec, 2021-01-28:
OK, thanks for the tip

oldman4U, 2021-01-28:
Please confirm that this also works for you.

Andy-ABTec, 2021-01-28:
Works for me as well, also, don't know if it's related but the TFT is not changing any of the files on the card to .cur's. 

For what its worth I haven't made any changes to my config.ini or configuration.h since mid-December last year, I've just been picking up bug fixes occasionally.

The only recent change I've made is in Marlin's configuration_adv.h, where I enabled HOST_PROMPT_SUPPORT as described in the TFT's config.ini

oldman4U, 2021-01-29:
Could you please try this firmware to see if it solves the issue with the crash while loading filament: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1eb47368ea26ee08b69ca7668ac8d4b55b0409e4

The firmware above is the last one which works for me before PR #1466 which causes a crash on my TFT when loading filament. I even did not know that the TFT can crash;-)

Please let me know

Andy-ABTec, 2021-01-29:
Package compiled OK, and copied to a newly formatted card. On boot the .bin is read then the tft tells me its found "Outdated data - Config: Update required" cannot get any further...

Im going to try a compiled version from the BTT Repo and see if that works...

oldman4U, 2021-01-29:
You also have to copy the config.ini and the TFT folder from the link above.

Andy-ABTec, 2021-01-29:
I did

Andy-ABTec, 2021-01-29:
Icopied the bin, config.ini the font file and the graphics

Andy-ABTec, 2021-01-29:
BTW the precompiled file is doing exactly the same thing - rebooting...

Andy-ABTec, 2021-01-29:
Could it be Marlin Bugfix 2.0.x on my BTT SKR Mini E3 that's causing the problem?

sarvenn, 2021-01-29:
> verify on Marlin fw you have all the following setting enabled:
>
> Options to support dialog with host:
> EMERGENCY_PARSER (in Configuration_adv.h)
> SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h) [...]

@digant73 should I undefine  #define CR10_STOCKDISPLAY and define //#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in configuration.h ? I checked the config file for Marlin bugfix skr mini e3 v2 and default is cr10_stockdisplay.

Andy-ABTec, 2021-01-29:
I've already done this, the only thing I was missing was HOST_PROMPT_SUPPORT which I've fixed

digant73, 2021-01-29:
@sarvenn no, if you have a creality printer

oldman4U, 2021-01-29:
README

oldman4U, 2021-01-29:
@Andy-ABTec now I also tried the precompiled from above and it works for me. Do you have a "Extrude" button or an (Un)Load?

I use Marlin bug fix from Jan. 28th with HOST_PROMPT_SUPPORT activated.

Please send a movie which shows the crash.

Andy-ABTec, 2021-01-29:
Will do, unfortunately, I can't right now, Octoprint is busy printing for the next few hours, but I'll put something together as soon as I can.

oldman4U, 2021-01-31:
Still printing?

The crash while loading filament happens during pausing a print as well as before a print is started.

Could you please try this firmware to see if it solves the issue with the crash while loading filament: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1eb47368ea26ee08b69ca7668ac8d4b55b0409e4

Andy-ABTec, 2021-01-31:
Apologies had a bit of a disaster with GitHub over the weekend - Successfully trashed all my saved firmware. Currently rebuilding from scratch. will get back when I've finished...

Andy-ABTec, 2021-02-07:
Sorry for the delay, I'm back up and running now, latest Marlin Bugfix and latest Touch Screen firmware... Suffering the occasional freeze especially with OctoPrint but the original issue seems to be gone. Unless there's anything else you'd like me to try (BTT SKR Mini E3 V2 + BTT TFT 35 E3 V3 + BLTouch 3.0) for the sake of helping out I'll drop out of this chat now - thanks for your time, and sorry I wasn't more help.

kisslorand, 2021-02-07:
If original issue is gone, please close this ticket.

Andy-ABTec, 2021-02-07:
Don't appear to have a comment and close button, only a comment button - not my thread originally. That's why I said what I said...

oldman4U, 2021-02-07:
Andy could you please test one thing for me. Heat the Nozzle and then go to (Un)Load. Press Load and once asked press Purge More and DO NOT press Continue. 

Let me know what happens please.

Thank you

Andy-ABTec, 2021-02-07:
Have a print that finishes in around 5 hours, you're next in the queue...

oldman4U, 2021-02-07:
You need more printer 😁

Thank you

kisslorand, 2021-02-07:
> Don't appear to have a comment and close button, only a comment button - not my thread originally. That's why I said what I said...

OK, my mistake. :)

Andy-ABTec, 2021-02-07:
Am currently printing bits for a HyperCube Evolution... 😎

oldman4U, 2021-02-07:
Good starting point.

Andy-ABTec, 2021-02-08:
Just done the "Load" - "Purge More" thing... Something restarts, I'm back at the BTT logo on the TFT 35, then see the usual messages about Fade Height etc as if the printer has just been turned on.

Not sure if it's just the TFT restarting of if the TFT is telling the SKR Mini to do the purge and the SKR Mini is causing the problem.

Are there logs anywhere that might help?

oldman4U, 2021-02-08:
It is all good. Thank you 

@kisslorand

I have been hunting the (Un)Load crash now for several days using SKR Pro, v1.4 and E3 DIP. Other users reporting the same issue use SKR minis and all points to an issue related to STM32. All boards with the crash have this kind of processor. LPC based boards like my SKR v1.4 using the absolute identical firmware do not crash.

sarvenn, 2021-02-13:
After enabling the feautres mentioned in the TFT config.ini file, persistant pop up problem is gone(only continue option). Now it askes for purge more and continue. However on purge more choice printer reboots (skr mini e3 v2 +tft35). I know it is already known issue.

oldman4U, 2021-02-27:
The issue with the crash/hang after selecting "Purge More" when using a STM32 based mainboard is likely fixed thanks to user arjanmels

MarlinFirmware/Marlin#21210

Please update Marlin and close the ticket in case you do not need it anymore.

oldman4U, 2021-04-08:
@Msq001

This Marlin related issue is fixed using the bugfix version of Marlin after February 27th and the user does not respond anymore. 

Thank you

closed 2021-04-08 by bigtreetech