bigtreetech/BIGTREETECH-TouchScreenFirmware #1565: [BUG] white lines after flashing mks tft28 firmware

SimonePanada, 2021-01-25:
I installed the latest tft version on my artillery sidewinder x1 which has a mks tft28 display. how can i solve it? i tried to format my sd card from a linux os and from a windows os with various file system types like fat, fat 32 and exfat, but obtaining the same result in the photo 
[image]

digant73, 2021-01-25:
Flash also icons and config.Ini

SimonePanada, 2021-01-25:
> Flash also icons and config.Ini
 config.ini file k
in the sd card i put bmp and font folder taken from tft28 folder, mkstft28.bin file and two language file (italian and english)

oldman4U, 2021-01-25:
The ReadMe shows the correct folder structure and I guess this is also true for non BTT TFT's.

radek8, 2021-01-25:
Nobody reads those instructions anyway :-(

oldman4U, 2021-01-25:
Maybe we should stop helping people where the problem is clearly not
firmware related and where the solution is well documented.

digant73, 2021-01-25:
You must must have "tft28" directory in the sd card

SimonePanada, 2021-01-25:
> The ReadMe shows the correct folder structure and I guess this is also true for non BTT TFT's.

ahaha also tried with the folder structure of the ReadMe file before you wrote this useless reply and if i'm here it is because it didn't work. 
 i tried every solution i found on web that it could work. some guys solved it putting the two folders in the main directory so i tried it.

SimonePanada, 2021-01-25:
> You must must have "tft28" directory in the sd card

tried it with no results

radek8, 2021-01-25:
And in the TFT28 directory, do you have a BMP folder with icons and a FONT folder with fonts?

SimonePanada, 2021-01-25:
> And in the TFT28 directory, do you have a BMP folder with icons and a FONT folder with fonts?

I copied the whole tft28 folder taken from github, so yes, there is everything  needed. 
i read abut changing sd card but i didn'try it yet because i could,'t find any other sd card under 8gb in my house. 
so i asked to some friends if they have some sd cards and when i will have them i will try.

radek8, 2021-01-25:
OK

radek8, 2021-01-25:
Hold only one language file on the card. English is in the precompiled BIN file by default

SimonePanada, 2021-01-26:
i downloaded the same firmware from here https://www.thingiverse.com/thing:4294049, it hase the same files but it compiled fine. i think there is a problem with the mkstft28 files on this repository

kisslorand, 2021-01-26:
For this repository BMP and  FONT folders should go into the MKS folder, as defined in the "pin_MKS_TFT32_V1_4.h" file:
```pin_MKS_TFT32_V1_4.h
#ifndef ROOT_DIR
   #define ROOT_DIR "MKS"
#endif```

oldman4U, 2021-01-26:
Maybe add this information to your PR #1557. This could be useful for users with a MKS display.

digant73, 2021-01-27:
@SimonePanada did you successfully flash the icons/fonts renaming folder "tft28" to "mks"?

SimonePanada, 2021-01-27:
> @SimonePanada did you successfully flash the icons/fonts renaming folder "tft28" to "mks"?

can't find any other microsd under 8 gb.. i'm asking literally to everyone if they have one. this morning i was digging into the problem and i found [here on github](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/676) that if i have a mks tft v3 i should use [this fork](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) and that i should use the mks tft32 firmware. i will try in a few hours.

digant73, 2021-01-27:
Try first this fw. You simply need to rename folder TFT28 to MKS and it should work

SimonePanada, 2021-01-27:
it worked! the problem was the name of the folder!

closed 2021-01-27 by SimonePanada