bigtreetech/BIGTREETECH-TouchScreenFirmware #1568: Mesh data not displayed correctly

RocksaltX, 2021-01-26:
Installed latest firmware for BIGTREETECH-TouchScreenFirmware-master on TFT 3.5 V2
Mesh data display is one row and not a full grid.
Flashing firmware does not remove old data of stored mesh data

### Steps to reproduce

1. Flash firmware
2. navigate to : menu, movement,  bed level, mesh data
3. grid is not displayed. Only one row of select able points exists.

**Expected behavior**
9X9 grid to appear

**Actual behavior**
See above

### Hardware Variant

TFT_35_V2_.0.26.x Jan 23 2021

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2021-01-26:
What type of leveling do you use?
Attach a photo of the problem and Marlin configuration files.

RocksaltX, 2021-01-26:
AUTO_BED_LEVELING_UBL. I have flashed back 1 rev and it's working. I would like the newer firmware to work. All functions work except for the mesh data grid only showing one vertical line and not a 9X9.

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5872386/Marlin.zip)

digant73, 2021-01-26:
From octoprint/pronterface type the command M420 V1 T1 and provide the output

RocksaltX, 2021-01-26:
0.290	0.220	0.180	0.130	0.070	0.060	0.060	0.080	0.040
0.180	0.110	0.070	0.030	0.010	0.020	0.020	0.020	0.010
0.130	0.060	0.040	0.020	0.010	0.020	0.020	0.020	0.030
0.090	0.040	0.020	0.020	0.000	0.010	0.020	0.020	0.040
0.080	0.020	0.000	0.000	-0.010	0.010	0.020	0.030	0.040
0.060	0.000	-0.020	-0.020	-0.020	0.000	0.020	0.030	0.040
0.060	-0.020	-0.040	-0.040	-0.030	-0.010	0.010	0.020	0.050
0.080	0.000	-0.040	-0.060	-0.040	-0.020	0.000	0.020	0.070
-0.050	0.020	-0.080	-0.050	-0.050	-0.020	-0.010	0.020	0.090

digant73, 2021-01-26:
the output for M420 V1 T1 should be more verbose and provide as first row the following string (for UBL):

Bed Topography Report for CSV:

Are you sure you executed M420 V1 T1 and not a less verbose command?

RocksaltX, 2021-01-26:
SENDING:M420 V1 T1
Bed Topography Report for CSV:
0.290	0.220	0.180	0.130	0.070	0.060	0.060	0.080	0.040
0.180	0.110	0.070	0.030	0.010	0.020	0.020	0.020	0.010
0.130	0.060	0.040	0.020	0.010	0.020	0.020	0.020	0.030
0.090	0.040	0.020	0.020	0.000	0.010	0.020	0.020	0.040
0.080	0.020	0.000	0.000	-0.010	0.010	0.020	0.030	0.040
0.060	0.000	-0.020	-0.020	-0.020	0.000	0.020	0.030	0.040
0.060	-0.020	-0.040	-0.040	-0.030	-0.010	0.010	0.020	0.050
0.080	0.000	-0.040	-0.060	-0.040	-0.020	0.000	0.020	0.070
-0.050	0.020	-0.080	-0.050	-0.050	-0.020	-0.010	0.020	0.090
Mesh is valid
Storage slot: 1
echo:Bed Leveling ON
echo:Fade Height 2.00
>>> m503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.7500
echo:; Steps per unit:
echo: M92 X80.0000 Y80.0000 Z400.0000 E372.0000
echo:; Maximum feedrates (units/s):
echo:  M203 X1200.0000 Y1200.0000 Z10.0000 E50.0000
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1200.0000 Y1200.0000 Z100.0000 E5000.0000
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.0000 R1000.0000 T1000.0000
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.0000 S0.0000 T0.0000 X6.0000 Y6.0000 Z0.3000 E5.0000
echo:; Home offset:
echo:  M206 X0.0000 Y0.0000 Z0.0000
echo:; Unified Bed Leveling:
echo:  M420 S1 Z2.0000
Unified Bed Leveling System v1.01 active
echo:; Active Mesh Slot: 1
echo:; EEPROM can hold 9 meshes.
echo:; Material heatup parameters:
echo:  M145 S0 H230 B80 F0
echo:  M145 S1 H270 B105 F0
echo:; PID settings:
echo:  M301 P19.4700 I1.5900 D59.4000
echo:  M304 P202.0000 I11.0000 D955.0000
echo:; Z-Probe Offset (mm):
echo:  M851 X-55.0000 Y0.0000 Z-2.7400
echo:; Stepper driver current:
echo:  M906 X707 Y707 Z500
echo:  M906 I1 Z500
echo:  M906 T0 E800
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 I1 Z
echo:; Linear Advance:
echo:  M900 K0.1300
echo:; Filament load/unload lengths:
echo:  M603 L0.0000 U100.0000
echo:; Backlash compensation:
echo:  M425 F1.0000 X0.0300 Y0.0200 Z0.0500

RocksaltX, 2021-01-26:
Ever since using UBL the mesh data using M420 V contains characters that are not present when using bed_leveling_bilinear. This is how it looked then using M420 V
     0      1      2      3      4      5      6
 0 -0.015 -0.098 -0.072 -0.075 -0.081 -0.067 -0.037
 1 -0.045 -0.122 -0.090 -0.091 -0.086 -0.075 -0.054
 2 -0.044 -0.118 -0.095 -0.103 -0.111 -0.108 -0.093
 3 -0.050 -0.116 -0.088 -0.093 -0.097 -0.090 -0.075
 4 -0.017 -0.087 -0.075 -0.100 -0.106 -0.106 -0.094
 5 -0.006 -0.077 -0.064 -0.074 -0.070 -0.062 -0.053
 6 +0.052 -0.033 -0.027 -0.051 -0.053 -0.054 -0.029

Now:
 0       1       2       3       4       5       6       7       8
 8 | +0.510  +0.380  +0.280  +0.160  +0.050  +0.020  +0.010  +0.030  -0.060
   |
 7 | +0.330  +0.200  +0.110  +0.030  -0.020  -0.010  -0.030  -0.040  -0.050
   |
 6 | +0.210  +0.120  +0.050  +0.050  +0.010   0.000   0.000  -0.020   0.000
   |
 5 | +0.160  +0.090  +0.070  +0.040  +0.010  +0.010  +0.010  +0.020  +0.030
   |
 4 | +0.130  +0.060 [+0.010]  0.000  -0.020  +0.010  +0.020  +0.010  +0.020
   |
 3 | +0.110  +0.020  -0.030  -0.020  -0.030  -0.010   0.000  +0.020  +0.040
   |
 2 | +0.100  -0.030  -0.060  -0.070  -0.060  -0.030  -0.010  -0.010  +0.030
   |
 1 | +0.110  -0.020  -0.080  -0.100  -0.100  -0.050  -0.050  -0.030  +0.040
   |
 0 | +0.110  +0.050  -0.070  -0.110  -0.110  -0.090  -0.090  -0.060  +0.070
        0       1       2       3       4       5       6       7       8
    ( 19, 19)                                                      (281, 19)

   0       1       2       3       4       5       6       7       8

RocksaltX, 2021-01-26:
I am guesing, the firmware will work with bed_leveling_bilinear. I will see if I have an old firmware file or compile another w/bilinear and see,

gordo3di, 2021-01-26:
I had the same issue. We've been testing our firmware a lot and I have changed every parameter I can think of but issue persists. At first with UBL it would display only a single row with only 4 points (almost like it was the corner leveling). I've tried this with a 3x3, 4x4, 10x10, 12x12 and 14x14 grid and I tried saving via M500 just in case.

Now with BBL it displays no data just white (I can still see the check mark and cancel icons etc). Previously this was working fine. The mesh is fine and I can output the data, also leveling is working as I print so It seems like the data is there.

I'm using the touchscreen firmware from 1/19/21 on a TFT35 V3 with SKR v1.4 Turbo. Using Marlin 2.0.6.1

```
![PXL_20210126_184547160](https://user-images.githubusercontent.com/11464057/105892050-53034100-5fdf-11eb-9c2f-cb7472cec9d2.jpg)

Connecting...
Printer is now online.
>>> M420 V1 T1
SENDING:M420 V1 T1
Bilinear Leveling Grid:
      0      1      2      3      4      5      6      7      8      9     10     11
 0 +0.495 +0.596 +0.872 +0.739 +0.744 +0.457 +0.215 -0.011 -0.283 -0.534 -0.780 -1.019
 1 +0.722 +0.714 +0.942 +0.759 +0.686 +0.356 +0.127 -0.070 -0.188 -0.344 -0.493 -0.709
 2 +0.942 +0.895 +1.017 +0.787 +0.704 +0.397 +0.151 -0.004 -0.125 -0.240 -0.300 -0.419
 3 +1.117 +1.102 +1.100 +0.892 +0.774 +0.475 +0.275 +0.129 +0.032 -0.063 -0.103 -0.140
 4 +1.226 +1.181 +1.176 +0.960 +0.840 +0.544 +0.371 +0.217 +0.100 -0.004 -0.014 +0.012
 5 +1.310 +1.234 +1.200 +0.990 +0.875 +0.599 +0.449 +0.292 +0.137 +0.027 +0.034 +0.077
 6 +1.291 +1.296 +1.281 +1.051 +0.857 +0.689 +0.474 +0.307 +0.189 +0.106 +0.094 +0.114
 7 +1.286 +1.341 +1.334 +1.156 +0.947 +0.714 +0.476 +0.317 +0.209 +0.151 +0.089 +0.054
 8 +1.095 +1.387 +1.350 +1.207 +1.031 +0.807 +0.525 +0.356 +0.211 +0.115 +0.015 -0.104
 9 +0.935 +1.281 +1.252 +1.200 +1.081 +0.902 +0.632 +0.396 +0.186 +0.017 -0.145 -0.366
10 +0.749 +1.085 +1.092 +1.142 +1.046 +0.872 +0.622 +0.349 +0.120 -0.110 -0.366 -0.599
11 +0.416 +0.760 +0.785 +0.942 +0.909 +0.795 +0.555 +0.234 -0.103 -0.414 -0.629 -0.804
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15       16       17       18       19       20       21       22
 0 +0.49500 +0.53469 +0.59625 +0.74906 +0.87250 +0.82258 +0.73875 +0.75078 +0.74375 +0.61609 +0.45750 +0.33250 +0.21500 +0.10367 -0.01125 -0.14531 -0.28250 -0.40969 -0.53375 -0.65766 -0.78000 -0.89984 -1.01875
 1 +0.60922 +0.61672 +0.65102 +0.79165 +0.90719 +0.84607 +0.74820 +0.73884 +0.71031 +0.56638 +0.39797 +0.27491 +0.16430 +0.05486 -0.04844 -0.14147 -0.23297 -0.33239 -0.43336 -0.52985 -0.63031 -0.74420 -0.86250
 2 +0.72250 +0.70328 +0.71375 +0.83906 +0.94250 +0.86945 +0.75875 +0.73164 +0.68625 +0.53102 +0.35625 +0.23359 +0.12750 +0.02180 -0.07000 -0.13133 -0.18750 -0.26367 -0.34375 -0.41438 -0.49250 -0.59641 -0.70875
 3 +0.83578 +0.80367 +0.79875 +0.89988 +0.97922 +0.88984 +0.76781 +0.73429 +0.68703 +0.53629 +0.36570 +0.23679 +0.12617 +0.03066 -0.04883 -0.10725 -0.16016 -0.22636 -0.29109 -0.33834 -0.39063 -0.47173 -0.56180
 4 +0.94250 +0.90812 +0.89500 +0.96766 +1.01750 +0.91539 +0.78750 +0.75039 +0.70375 +0.56078 +0.39750 +0.26492 +0.15125 +0.06594 -0.00375 -0.06688 -0.12500 -0.18633 -0.24000 -0.26977 -0.30000 -0.35570 -0.41875
 5 +1.03695 +1.01572 +1.00516 +1.04374 +1.05867 +0.95784 +0.83758 +0.79165 +0.73570 +0.59287 +0.43453 +0.31196 +0.20859 +0.12745 +0.06109 +0.00441 -0.04656 -0.10133 -0.14844 -0.17322 -0.19477 -0.23092 -0.27078
 6 +1.11750 +1.10922 +1.10250 +1.11328 +1.10000 +1.00352 +0.89250 +0.83883 +0.77375 +0.62945 +0.47500 +0.36547 +0.27500 +0.19539 +0.12875 +0.07742 +0.03250 -0.01852 -0.06250 -0.08609 -0.10250 -0.12141 -0.14000
 7 +1.17758 +1.16355 +1.15156 +1.15823 +1.14180 +1.04337 +0.93094 +0.87546 +0.80906 +0.66385 +0.51078 +0.40909 +0.32602 +0.24625 +0.17672 +0.12201 +0.07375 +0.01999 -0.02398 -0.04237 -0.04875 -0.05102 -0.05039
 8 +1.22625 +1.20125 +1.18125 +1.18945 +1.17625 +1.07531 +0.96000 +0.90500 +0.84000 +0.69516 +0.54375 +0.44859 +0.37125 +0.29094 +0.21750 +0.15562 +0.10000 +0.04141 -0.00375 -0.01688 -0.01375 -0.00289 +0.01250
 9 +1.27609 +1.23937 +1.20852 +1.20722 +1.18781 +1.08735 +0.97539 +0.92409 +0.86273 +0.71900 +0.56992 +0.48356 +0.41445 +0.33605 +0.25961 +0.18683 +0.11977 +0.05638 +0.01062 +0.00186 +0.01180 +0.02958 +0.05227
10 +1.31000 +1.26922 +1.23375 +1.22523 +1.20000 +1.10008 +0.99000 +0.93664 +0.87500 +0.73906 +0.59875 +0.51625 +0.44875 +0.37094 +0.29250 +0.21211 +0.13750 +0.07242 +0.02750 +0.02102 +0.03375 +0.05328 +0.07750
11 +1.30617 +1.28494 +1.26547 +1.26354 +1.23883 +1.13529 +1.01594 +0.93902 +0.86281 +0.75588 +0.64562 +0.55232 +0.46594 +0.38251 +0.30406 +0.23016 +0.16422 +0.10631 +0.06602 +0.05811 +0.06703 +0.08302 +0.10344
12 +1.29125 +1.29500 +1.29625 +1.30344 +1.28125 +1.17742 +1.05125 +0.95055 +0.85750 +0.77445 +0.68875 +0.58109 +0.47375 +0.38461 +0.30750 +0.24289 +0.18875 +0.14086 +0.10625 +0.09359 +0.09375 +0.10172 +0.11375
13 +1.29953 +1.31143 +1.31977 +1.32988 +1.31156 +1.22047 +1.10437 +0.99952 +0.89617 +0.79980 +0.70102 +0.58522 +0.47352 +0.38487 +0.31102 +0.25037 +0.20180 +0.16431 +0.13594 +0.11389 +0.09961 +0.09570 +0.09586
14 +1.28625 +1.31766 +1.34125 +1.35203 +1.33375 +1.25758 +1.15625 +1.05539 +0.94750 +0.83242 +0.71375 +0.59031 +0.47625 +0.38883 +0.31750 +0.25680 +0.20875 +0.17711 +0.15125 +0.11859 +0.08875 +0.06953 +0.05375
[... cut ...]
20 +0.74875 +0.93742 +1.08500 +1.10664 +1.09250 +1.12398 +1.14250 +1.10836 +1.04625 +0.96898 +0.87250 +0.75375 +0.62250 +0.48430 +0.34875 +0.23164 +0.12000 +0.00672 -0.11000 -0.23797 -0.36625 -0.48398 -0.59875
21 +0.59164 +0.78119 +0.93055 +0.95397 +0.94797 +1.00499 +1.05141 +1.03332 +0.98391 +0.92137 +0.83672 +0.72389 +0.59234 +0.44594 +0.29547 +0.15527 +0.01852 -0.11823 -0.25086 -0.37672 -0.49492 -0.60120 -0.70297
22 +0.41625 +0.60805 +0.76000 +0.78414 +0.78500 +0.86742 +0.94250 +0.94258 +0.90875 +0.86477 +0.79500 +0.68797 +0.55500 +0.40039 +0.23375 +0.06500 -0.10250 -0.26570 -0.41375 -0.52977 -0.62875 -0.71875 -0.80375
echo:Bed Leveling ON
echo:Fade Height OFF
```

RocksaltX, 2021-01-26:
OK... the LCD visual grid works properly with bed_leveling_bilinear. 
SENDING:M420 V1 T1
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 +0.432 +0.311 +0.025 -0.085 -0.132 -0.133 -0.128
 1 +0.504 +0.165 -0.039 -0.118 -0.186 -0.198 -0.217
 2 +0.407 +0.131 -0.038 -0.113 -0.198 -0.213 -0.254
 3 +0.400 +0.189 -0.003 -0.087 -0.176 -0.210 -0.260
 4 +0.432 +0.233 +0.081 -0.005 -0.126 -0.167 -0.243
 5 +0.560 +0.340 +0.154 +0.063 -0.034 -0.090 -0.164
 6 +0.807 +0.591 +0.417 +0.226 +0.110 +0.072 +0.064

Here is what it looks like giving M420 V with UBL
SENDING:M420V
Bed Topography Report:
    ( 12,288)                                                      (288,288)
        0       1       2       3       4       5       6       7       8
 8 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 7 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 6 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 5 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 4 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 3 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 2 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 1 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
   |
 0 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
        0       1       2       3       4       5       6       7       8
    ( 12, 12)                                                      (288, 12)
Mesh is valid

Here is what it looks like with: M420 V1 T1
SENDING:M420 V1 T1
Bed Topography Report for CSV:
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
Mesh is valid

RocksaltX, 2021-01-26:
Also, I tried marlin bugfix 2.0.x and marlin 2,.0. What is interesting is that the previous version of BTT35V2 works properly with UBL and mesh display.

digant73, 2021-01-26:
So with previous tft fw and same marlin fw it was ok? If so I suspect that the output from M420 V1 T1 is no more fully intercepted by mesh editor. I will install the latest tft fw and chek

RocksaltX, 2021-01-26:
Correct. 

Also, the main LCD reports/displays the mesh table correctly for manual mesh editing and UBL. I am also using REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER along with the TFT35V2

digant73, 2021-01-26:
Found the problem.
The PR #1487 has overwritten the standard function "strtod" (used in Mesh Editor) with a custom function in file my_mysc.c.
@guruathwal can you fix it or restore the standard function? the behaviuor has to be has the documented standard function strtod.
Basically the next pointer is never changed and moved to the next token to parse (so it's one shot)

guruathwal, 2021-01-27:
@digant73 can you share the actual string/text not parsing correctly?

digant73, 2021-01-27:
not possible now unfortunately (I'm at office).
Can you verify the result in mesh editor in your printer with your strtod and the original strtod? probably you will get the same results. In my printer I got a matrix with only the first column (a 1x10 instead of 10x10).

guruathwal, 2021-01-27:
@digant73 found the problem and fixed as you suspected.

RocksaltX, 2021-01-28:
Thank you so much. Working as it should now.

closed 2021-01-30 by bigtreetech

gordo3di, 2021-01-31:
Ok I've tested this a bit. It looks like the mesh grid doesn't display if you have bilinear subdivision enabled. I have a 12x12 grid with bilinear subdivision enabled at 2 and it doesn't work... Turn BS off and its fine... Just putting it here so people know.

digant73, 2021-02-01:
@gordo3di you should continue to see a 12x12 grid and not the BS right?
If so, it's correct. The process is stopped when trying to parse a row not containing 12 numbers. So in your case when the string "Subdivided with CATMULL ROM Leveling Grid:" is parsed

gordo3di, 2021-02-08:
> @gordo3di you should continue to see a 12x12 grid and not the BS right?

Nope. With BS on I see nothing just a white screen (as in the pic above). As soon as I turn off BS and recompile it displays as it should.

digant73, 2021-02-08:
@gordo3d starnge you should see the 12x12 grid.
Can you share the changes you made to enable BS? so I will try to replicate the issue

gordo3di, 2021-02-08:
I'm Using Marlin 2.0.6.1 and it's been tailored for the gMax 2. For bed leveling I'm using BBL and the only changes was to turn bilinear subdivision on or off then reset the printer and re-level. 

```
#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  //gMax 2 Pro
  #define GRID_MAX_POINTS_X 12
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  // Probe along the Y axis, advancing X after each column
  #define PROBE_Y_FIRST

  #if ENABLED(AUTO_BED_LEVELING_BILINEAR)

    // Beyond the probed grid, continue the implied tilt?
    // Default is to maintain the height of the nearest edge.
    #define EXTRAPOLATE_BEYOND_GRID

    //
    // Experimental Subdivision of the grid by Catmull-Rom method.
    // Synthesizes intermediate points to produce a more detailed mesh.
    //
    //#define ABL_BILINEAR_SUBDIVISION
    #if ENABLED(ABL_BILINEAR_SUBDIVISION)
      // Number of subdivisions between probe points
      #define BILINEAR_SUBDIVISIONS 2
    #endif

  #endif
```

PR #1558 (2021-01-30): Reduce global variables and cleanup
- Reduce/remove global variable to reduce RAM usage.
- Fix Busy popup preventing the user from exiting load-unload menu if machine/printer is busy or not accepting commands.
- `stringToDouble` now ignores tab character. (UBL mesh data reported by printer uses tab character instead of spaces causing only first value to be parsed) #1568 

Fixes #1568