bigtreetech/BIGTREETECH-TouchScreenFirmware #1569: skr 1.3 filament sensor TFT24_V1.1 marlin

wolfam0108, 2021-01-26:
Hey. Faced a problem.
A filament sensor is connected to the skr 1.3 motherboard. To contact Xmin. The filo sensor is a simple limit switch.

When I use the TFT24 in LCD12864 emulation mode, the sensor is triggered and the plastic replacement sequence starts.
But in touch mode when printing from USB, printing does not stop when the sensor is triggered.

- marlin 2.0.7.2
in the config Configuration_adv.h are active:
M115_GEOMETRY_REPORT
M114_DETAIL
REPORT_FAN_CHANGE
EMERGENCY_PARSER
SERIAL_FLOAT_PRECISION 4
HOST_ACTION_COMMANDS
AUTO_REPORT_TEMPERATURES
AUTO_REPORT_SD_STATUS
LONG_FILENAME_HOST_SUPPORT
FILAMENT_LOAD_UNLOAD_GCODES

in Configuration.h are active
SDSUPPORT

In the config.ini I tried states 0, 1.
#### Default Filament Runout Sensor
# Options: [NONE: 0, NORMAL: 1, SMART: 2]
fil_runout: 0

Who is facing a similar problem? Is it possible to implement such a scheme for the filoment sensor?

oldman4U, 2021-01-26:
Please read the list of verified bugs before opening a new ticket.

Thank you

wolfam0108, 2021-01-26:
Please tell me what topic was the answer?

oldman4U, 2021-01-26:
What do you mean?

kisslorand, 2021-01-31:
Put M75 in your start gCode or at the beginning of your gCode itself.

closed 2021-04-08 by stale[bot]