bigtreetech/BIGTREETECH-TouchScreenFirmware #1570: Marlin Mode error with MKS SGen L V1.0

luwi66, 2021-01-26:
When I connect TFT35 E3 V3.0 with cables EXP1 and EXP2 to motherboard MKS SGen L V1.0, I get problems.
The motherboard has Marlin-Bugfix 2.0x 
Touch mode  is working fine, when EXP1 and EXP2 are not connected to  the MKS board.
When connected, there are two cases
1) Cables connected twisted (correct connection as decribed in BIGTREETECH-TouchScreenFirmware):
    screen is flashing grey-black about twice per second
2) Cables connected not-twisted:
    a loud tone; BTT logo and then both mode icons (LCD12864 mode and Touch mode); Now neither touching nor knob is 
    working.
Without the cables connected, Touch mode is working fine.

When I connect the MKS board to an old REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, Marlin is working ok. So the graphic controller definition in Marlin's configuration.h is ok.

radek8, 2021-01-26:
how do i understand " Cables connected twisted "?
Can you take a picture of how connected you are?

radek8, 2021-01-26:
look at
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1460
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1001

luwi66, 2021-01-27:
By twisted I mean reversed; the old known problem with the MKS Gen boards and their EXP cables.In my post I have adressed both directions, so a picture will not help. Gesendet von Mail für Windows 10 Von: RadekGesendet: Dienstag, 26. Januar 2021 18:13An: bigtreetech/BIGTREETECH-TouchScreenFirmwareCc: luwi66; AuthorBetreff: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] Marlin Mode error with MKS SGen L V1.0 (#1570) how do i understand " Cables connected twisted "?Can you take a picture of how connected you are?

luwi66, 2021-01-27:
In both threads my error condition does not come up, so they don’t help.The only issue new for me was one comment, where it is said, that you don’t need 5V and Gnd from the RS232 cable when EXP cables are connected.I tried this, but no effect to my Problem. Gesendet von Mail für Windows 10 Von: RadekGesendet: Dienstag, 26. Januar 2021 18:18An: bigtreetech/BIGTREETECH-TouchScreenFirmwareCc: luwi66; AuthorBetreff: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] Marlin Mode error with MKS SGen L V1.0 (#1570) look at#1460#1001

radek8, 2021-01-27:
Disconnect the RS232 cable
Connect cables EXP1 and EXP2
Does the display light up?

radek8, 2021-01-27:
one picture is more than 1000 words

luwi66, 2021-01-27:
Sorry, seems the TFT is gone. I can’t get it to light up anymore. Maybe the many tries with different cables and combinations have killed it Gesendet von Mail für Windows 10 Von: RadekGesendet: Mittwoch, 27. Januar 2021 14:19An: bigtreetech/BIGTREETECH-TouchScreenFirmwareCc: luwi66; AuthorBetreff: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] Marlin Mode error with MKS SGen L V1.0 (#1570) Disconnect the RS232 cableConnect cables EXP1 and EXP2Does the display light up?

luwi66, 2021-01-28:
Hi RadekThe setup is working now. After completely dismantling the TFT35 from ist housing and connecting to the board with a different set of cables; and giving the MKS mainboard a new firmware (Marlin 2.0.7.2), everything works fine. I cannot say where I produced the chaos earlier.I’m sorry for creating uneasiness and hope I didn’t steal too much from your time.Thanks for helping.Best regards Ludwig  Gesendet von Mail für Windows 10 Von: RadekGesendet: Mittwoch, 27. Januar 2021 14:21An: bigtreetech/BIGTREETECH-TouchScreenFirmwareCc: luwi66; AuthorBetreff: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] Marlin Mode error with MKS SGen L V1.0 (#1570) one picture is more than 1000 words

closed 2021-01-28 by luwi66