bigtreetech/BIGTREETECH-TouchScreenFirmware #1571: [BUG] (screen freezes on filament load)

sinole1987, 2021-01-27:
So I know that there are previous posts about how M0,M1 doesn't work on TFT screen. I have updated my screen firmware  recently, but I still have issues with loading filament. this time issue is worst. It used to be that screen fell into a error loop. now it freezes altogether.

1. heat up the nozzle.
2. press load.
3. wait for extruder to purge.
4. and then pop up apears. "printer paused" " waiting for user to respond"
5.press ok

 then screen freezes 

### Hardware Variant
using a SKR 1.4 turbo
TFT 43 connected through both marlin and serial port

oldman4U, 2021-01-27:
Kowing that there is an issue, what Is the difference to the other tickets?

sarvenn, 2021-01-28:
#1564 check this one @sinole1987 
I also have the same issue.
Have you compiled your own Marlin fw for your board or do you use the precompiled firmware?

oldman4U, 2021-01-28:
The screen freezes or you get the same popup again?

sinole1987, 2021-01-28:
> Kowing that there is an issue, what Is the difference to the other tickets?

the difference is screen freezing that didn't happen before. before it was just an error loop.

sinole1987, 2021-01-28:
ok I will, it is a year old ticket, but sure. and I didn't compile my own firmware. I can do some programing, but if I was proficient enough, I would have just made the code ignore the screen filament sensor and read the onboard one. it seems that feature is messing up the whole code. Can we have a firmware that works only with onboard filament sensor.

oldman4U, 2021-01-28:
I started VSC for the first time 1.5 years ago without any background in sw
development. It is where the fun starts and compiling your own firmware is
mandatory for being able to use BTT hardware.

sinole1987, 2021-01-28:
yes, i think at some point I will do that. i think marlin is still not very compatible with tft yet.

sarvenn, 2021-01-29:
> I started VSC for the first time 1.5 years ago without any background in sw development. It is where the fun starts and compiling your own firmware is mandatory for being able to use BTT hardware. sinole1987 <notifications@github.com> schrieb am Do. 28. Jän. 2021 um 17:44:
> […](#)

I've come to that point as well. I need to learn how to compile my own fw. Otherwise most of the problems I've seen will be unsolved because our printers are different. A common way can not be found. I will invest on myselft this weekend :)

oldman4U, 2021-01-29:
Keep in mind how many different hardware devices are available and this only includes no optional devices like filament runout and levelling items- This alone would end up in hundreds if not thousands of different combinations and therefore firmware versions. Some users have problems to find the right firmware for their TFT, even there are 3 handful of files to choose from. In case you want more than A4988, 1 extruder and a heated bed, you have to compile by yourself.

oldman4U, 2021-01-29:
@sinole1987

Could you please try this firmware to see if it solves the issue: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1eb47368ea26ee08b69ca7668ac8d4b55b0409e4

@kisslorand The reported issue is related to your PR as the release right before your PR (see link above) does not have the issue:
ExtrusionMenu-Load/UnLoadToMoreMenu #1466

Thank you

sinole1987, 2021-01-30:
yeh i can make basic, firmware or change a thing or two on the firmware, but not really edit subroutines and stuff. but i will try this version and let you know soon.

oldman4U, 2021-01-30:
There is nothing you have to change because the precompiled version already does the job.

sinole1987, 2021-01-31:
It doesnt freez anymore, but still loop until I use the knob to control the massage in marlin. thank you. this is better than freezing.

oldman4U, 2021-01-31:
You mean when you pause the print?

The issue exists also without pausing a print. It is enough to preheat the nozzle and to try to Load filament. This freezes or crashes the screen.

sinole1987, 2021-01-31:
yes, its all related to M0 M1 commands not working on TFT mode.

oldman4U, 2021-02-01:
sinole1987

You have to enable HOST_PROMPT_SUPPORT in Marlin configuration.adv to get the correct buttons (Purge More and Continue).

sinole1987, 2021-02-01:
oh yes thank you. and I installed the firmware updated 2 days ago, seems be working fine. do still need to connect filament sensor to the screen as well if I want to have filament runout in TFT mode?

oldman4U, 2021-02-01:
sinole1987 and sarvenn

Can you both purge filament using Menu-(Un)Load more than once?

So Preheat the nozzle. Once heated go to Menu (Un)Load. Press "Load" and once you get the PurgeMore or Continue button press PurgeMore. What happens, and which mainboards do you have?

sinole1987, 2021-02-02:
ye purge more has issues for sure in new version. does the version you send me work fine ? or same problems?

oldman4U, 2021-02-02:
The version I sent has Extrude instead of (Un)Load which works for me.

kisslorand, 2021-02-07:
Finally I was able to reproduce this bug. It's very hard to trace the issue, I have to play 20~30 minutes to have the hiccup once on my printer. So far I saw that the TFT is the culprit, it's not frozen but waits buttons from another menu that is displayed.
I am monitoring the serial on both sides, motherboard and TFT, so far I didn't see any issue on any side.

oldman4U, 2021-02-08:
I have a several hours print going on on my machine, but I believe there could be a Marlin fix for the crash we see when the Purge More button is pressed. Here is the version which includes probably the fix for those who would like to check: 

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

sinole1987, 2021-02-08:
I kinda noticed that it might be a marlin issue reading all of the different threads and reviewing the code.

oldman4U, 2021-02-09:
Could someone please try kisslorand's possible fix: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1607

I will be able to test it in a few hours.

What I know so far is, that it affects STM32 (E3 mini, E3 DIP, Pro) boards but not LPC based motherboards (SKR v1.x) and that the problem always existed (tested firmware back to November last year). The problem became more obvious the time kisslorand moved the (Un)Load to the main screen, replacing the Extrude menu which does not have this problem.

sinole1987, 2021-02-10:
tested it doesn't work on LPC platform.

sinole1987, 2021-02-10:
Well, the way lpc motherboards connect for marlin mode is different from mini baords. we need two cables, since this is a communication issue. it makes sense. the pin mapping is wrong on the program. again I wish I could trace it myself.

oldman4U, 2021-02-10:
Which LPC based board have you tested and what is the problem exactly?

I thought the same first but why do I see the same issue using the SKR Pro v.1.1 which uses EXP 1 and 2?

sinole1987, 2021-02-11:
LPC 1769, and you are on track. All of these motherboards use EXP1 and 2, I am assuming the fix works on motherboards that use EXP3.

oldman4U, 2021-02-11:
Initially I also believed it has to do with the EXP connection. But the SKR
PRO destroyed this.

sinole1987, 2021-02-13:
I am actually happy with the filament detect sensor right now, you don't really need load button. only problem is that nozzle goes on top of the print and dumps a blub of plastic before continuing the print. if it does the dump on the bed and then goes to print its gonna be perfect. should that be arranged on firmware?

oldman4U, 2021-02-14:
All users can use the Extrude menu instead of (Un)Load without any problem as it does not use M701. Users with STM32 based mainboards can not use (Un)Load because it will crash the mainboard. This crash is not TFT firmware related and can be reproduced also without a TFT.

closed 2021-02-24 by bigtreetech

oldman4U, 2021-02-27:
The issue is likely fixed thanks to user arjanmels

https://github.com/MarlinFirmware/Marlin/pull/21210

Please update Marlin to see if it solves the problem.

PR #1607 (2021-02-24): Filament info for multi-filament print, possible fix for Load/Unload hanging popup, fix TFT freeze
### Requirements

BTT TFT or MKS TFT

### Description

This PR adds support for multi-filament prints. It enhances the print summary so it shows the sum of each  filament data present in the gCode. (sum of length, sum of weight, sum of cost)

Later on added a possible fix for the Load/Unload popup hanging issue.
...and more later added a fix for TFT freeze.

### Benefits

It benefits those who use the filament data embedded into the gCode and use multi-filament print.

### Related Issues

Impleme [... cut]