bigtreetech/BIGTREETECH-TouchScreenFirmware #1572: [FR] Optimized Print menu and List/thumbnail selection addition to top row

sarvenn, 2021-01-27:
**Is your feature request related to a problem? Please describe.**
Current thumbnail view only shows 5 boxes (files/folders). Up and down buttons doesn't shot up/down but left right and they are too big, back button as well. So these buttons may be optimized and placed on the right column just like "feature" menu.
This will let us to see 6 boxes at this page.
Other suggestion will let us to switch between list and thumbnail view quickly without the need to go to feature menu to change the view selection.

**Describe the solution you'd like**
See the picture
[image]

radek8, 2021-01-30:
It seems like a good idea
this would unify the file selection menu for printing
Just place the button at the top to switch the mode in the upper right corner.

radek8, 2021-01-30:
@kisslorand , should you take the time and mood to handle this?

sarvenn, 2021-01-30:
> unified

Thanks @radek8. I would be happy if this attracts a developer.
Let me correct a point.
In print selection screen, temperature values (if selected persistant temp reading is on in feature menu) are shown on the upper right corner if thumnail view is enabled, temp values can not be seen at file list view, that's why I placed the file/thumbnail selection button in the upper middle section. That's another open point I discovered now :) why not showing temp values in file list view option? Any logic behind? If not, then it is also something to be improved by the way.

sarvenn, 2021-01-30:
[image]
[image]
[image]

radek8, 2021-01-30:
You are right, in one window the temperature icons will not appear in the other yes. This window was just forgotten :-).
Another switching option would be to briefly hold down the "up arrow" or "down arrow" button

kisslorand, 2021-01-30:
>  Any logic behind? 

Yes, there is a logic to that, in fact there's an explanation.
The two types of files display use two different types of defined menu type. One is using `MENU_TYPE_ICON` and the other `MENU_TYPE_LISTVIEW'. 
`MENU_TYPE_ICON` has a title bar zone defined and all menus of this type will display the temperatures if "Persistent Temperature Info" is enabled.

Regarding the idea of 6 icons and navigation buttons on the right...
Implementing this needs a lot of work because the menu type would fall off the `MENU_TYPE_ICON` definition. Again, the amount of work needed outweighs the benefit of having one more file displayed.

sarvenn, 2021-01-30:
I think I understood the point.
My idea is something similar to combine the list and icon view somehow. So there would be a significant work load to combine them, defining the touch zones etc., right?
What if someone take his/her time and doesn't rush to do this? Technically, could it be achievable?

kisslorand, 2021-01-30:
> Technically, could it be achievable?

Yes, absolutely!

sarvenn, 2021-01-31:
So is it possible to add this idea to FR list. Maybe it attracts a developer who is not in hurry to do this. It may take too much effort but I believe it would worth it. I use the thumbnail view most of the time and those big buttons too bad for smaller screens like tft35.

oldman4U, 2021-01-31:
It is already.

oldman4U, 2021-02-05:
Please help the community and close the ticket once you do not need it anymore. Thank you

sarvenn, 2021-02-05:
I hope a developer will take this into his/her backlog to do in the near future :)

closed 2021-02-05 by sarvenn

sarvenn, 2021-03-05:
@oldman4U , @bigtreetech  should I re-open this and wait for a PR and merge to close it or should it stay closed since the item is added to feature list? #1170

oldman4U, 2021-03-05:
PRs can be merged completely independent of the existence of a ticket. It also does not matter if it is opened or closed.

bttguy, 2021-08-31:
+1 from me, would be very cool