bigtreetech/BIGTREETECH-TouchScreenFirmware #1573: [BUG] TFT35 wrong temp reading in PID

Duffy77, 2021-01-27:
### Description
When the Board is configured to use pidtempbed in Marlin, the reading is stopped when pid is reached.
e.g. you want to set the bed temp to 60 degrease, the reading is stopped when pid mode is reached and display is stucked at 50 degrease. 
In Octoprint i can still see the bed is heating and the temperature is going up.
After a few minutes the Firmware kills the process with a heating error.

### Steps to reproduce

1. Use an actual Marlin firmware where pidtempbed is enabled
2. Use the Touch Display to heat the printer (preheat pla)

**Expected behavior**
That the temparature will reach 200 and 60 degrease

**Actual behavior**
Nozzle temp increase till it reaches 200 degrease (correct)
Bed temp increase till 50 degrease and then no update is shown in display. 
After a few minutes an heating error occurs.

### Hardware Variant
TFT35 E3 V3.0 
### TFT Firmware Version & Main Board Firmware details

BIGTREE_TFT35_V3.0_E3.26.x.bin

SKR Mini E3 V2 
Firmware : BIGTREETECH-SKR-mini-E3/firmware/V2.0/Marlin-2.0.7.2-SKR-mini-E3-V2.0/
Self compiled with pidtempbed enabled

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

digant73, 2021-01-27:
did you update Marlin fw to latest bugfix version? With previous Marlin 2.0.7.2 I had a similar problem (the PID process failed).
That bug was fixed on latest Marlin fw bugfix version

Duffy77, 2021-01-27:
No i used the BTT Repo. this seems to be the problem..

Seems to be a Marlin firmware and not a tft firmware problem

digant73, 2021-01-27:
so, it is what I said. It is a bug on Marlin fw. Try to update Marlin fw to latest bugfix version

Duffy77, 2021-01-27:
Problem is in Marlin firmware and not in Display firmware.
Fixed in Marlin Bugfix Branch

closed 2021-01-27 by Duffy77