bigtreetech/BIGTREETECH-TouchScreenFirmware #1576: January 27 update tft35

Jeripimo, 2021-01-28:
X Y and Z position not updated during print

oldman4U, 2021-01-28:
Marlin emulation I guess.

But it has been working before using which firmware version?

Jeripimo, 2021-01-28:
It was working before. version TFT35_E3_V3.0.26.x Sep 12 2020
Using this display mode: https://www.reddit.com/r/BIGTREETECH/comments/iphojt/filament_cooling_fan_dont_show_off_on_tft35_e3/

And: fan speed not updating stays 0%

radek8, 2021-01-28:
> 
> And: fan speed not updating stays 0%

Many times it has already been addressed here.
Read the comments in Config.ini
In Marlin you have to enable the option
REPORT_FAN_CHANGE (in Configuration_adv.h)

oldman4U, 2021-01-29:
This information is several months old. Updating and reading will solve your problems. And I still don't know if you talk about Marlin or Touchscreen mode.

Jeripimo, 2021-01-29:
The problem is in touch screen mode using octoprint.
If I enable REPORT_FAN_CHANGE still no fan speed update in touch screen mode.

If printing from sd card in touch screen mode the fan speed is updated.

The internet example was just to show the screen mode I am using, touch screen.
In this mode x y and z position not updated.

Thanks

oldman4U, 2021-01-29:
In Touchscreen mode, there is no x,y and z shown during print. So I still have no idea what you mean, sorry.

Jeripimo, 2021-01-29:
If you look at the example in this link you can see the x y and z position
It's the same screen as on my tft35
https://www.reddit.com/r/BIGTREETECH/comments/iphojt/filament_cooling_fan_dont_show_off_on_tft35_e3/

oldman4U, 2021-01-29:
Yes. But this is the pre print screen. In this case the information that
you print using Octoprint could somehow be important.

In this case I would not answer you because I can not help.

kisslorand, 2021-01-29:
@Jeripimo 
 Do you have M114_DETAIL enabled in Configuration_adv.h?

Jeripimo, 2021-01-29:
Do I need to?
Version I am running now no issue only the latest one has that issue

kisslorand, 2021-01-29:
In the status screen the coordinates are queried every two seconds. For some reason Marlin doesn't answer. Enabling M114_DETAIL wouldn't hurt.
The conditions for the coordinate query are:
* printer connected to the TFT
* printer not in "wait" state
* a previous coordinate query got already an answer from Marlin

Jeripimo, 2021-01-29:
Ok I'll try that

kisslorand, 2021-01-31:
So?

Jeripimo, 2021-02-02:
Tried it today.

When enabling M114_DETAIL  my TFT won't connect to the printer even though the baud rate is correct.
Reflashed without M114 and it's ok again

kisslorand, 2021-02-03:
This is very odd. You should compare your Marlin config files against someone else's working ones.

oldman4U, 2021-02-07:
Please help the community and close the ticket once you do not need it anymore. Thank you

Jeripimo, 2021-02-08:
Not solved

closed 2021-02-08 by Jeripimo