bigtreetech/BIGTREETECH-TouchScreenFirmware #1577: [BUG] Unable to update firmware TFT35 E3 V3

fr0stsp1re, 2021-01-28:
### Description

Unable to update firmware

### Steps to reproduce

1. Load firmware(BIGTREE_TFT35_V3.0_E3.26.x.bin), icons and config file on  SDCARD formatted in FAT32
2. Insert SDCARD into TTFT35 and power on.
3. Push reset button.
4. Power cycle the display again.

**Expected behavior**
TFT35 E3 V3 should go into firmware update mode.

**Actual behavior**
TFT35 does not go into firmware update mode and boots directly to user interface.

### Hardware Variant
TFT35 E3 V3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.26.x.bin

I have also tried renaming the firmware per suggestions in this thread:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/24

radek8, 2021-01-28:
Many times it has already been addressed here.
Read the comments in Config.ini
In Marlin you have to enable the option
  REPORT_FAN_CHANGE (in Configuration_adv.h)

oldman4U, 2021-01-29:
Size of the sd card is? Tried another sd card? Followed the instructions in ReadMe?

fr0stsp1re, 2021-01-30:
> Size of the sd card is? Tried another sd card? Followed the instructions in ReadMe?

4Gb. Tried several SD cards and yes I followed the instructions in ReadMe. Seems like this is an issue covered in thread #24 however that thread seems to be for the non E3 board.

closed 2021-01-30 by fr0stsp1re