bigtreetech/BIGTREETECH-TouchScreenFirmware #1578: [Q] Ender3 - MKS Gen L v1 - BBT TFT35 3.0 - "No Printer Attached"

Silverender, 2021-01-29:
Hello, 

printer was working perfectly with Marlin 1.1.9 and firmware 3.0.25 on the screen but after updating screen to 3.0.26 I totally lost connection between printer and screen, "No printer Attached" message is always shown. 

After that I decided to update Marlin to 2.0.7.2 but still doesn't work, connection is just done by 4 cables RS232 for touch mode only, "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" is activated in Configuration.h and "CR10_STOCKDISPLAY" Deactivated. Baudrate on screen and marlin is the same (250000) also tried 115200 and didn't work either. 

Recompiled BTT TFT35 firmware from last master many times to avoid prebuilded releases but still doesn't work

This is really driving me crazy, any clue? All was working until screen update. Also tried to go back to 1.1.9 but doesn't work either.

Printer: Ender 3
Motherboard: MKS GEN L 1.0
Marlin: 2.0.7.2
Screen: BTT TFT35 3.0 - v.3.0.26 firmware.
Connection thru RS232 only

radek8, 2021-01-29:
If I'm not mistaken, you should be fine serial port -1
therefore
#define SERIAL_PORT -1

Silverender, 2021-01-29:
Hello radek,
When i tried -1 i can't compile, already tried shows error

radek8, 2021-01-29:
Pardon,
0 was correct
But nothing must be connected to the MKS USB port at the same time

https://github.com/MarlinFirmware/Marlin/issues/16408

radek8, 2021-01-29:
Uncomment
#define M114_DETAIL
#define REPORT_FAN_CHANGE

Silverender, 2021-01-29:
Will try, regarding usb port just and extension is connected i actually
have to unplug screen to flash or connect thru usb or computer is unable to
connect

radek8, 2021-01-29:
A USB port and a TFT port shared by the same port can only be connected by one of them at a time

Silverender, 2021-01-29:
Thanks, will try activating those lines and keep you posted about results

Silverender, 2021-02-01:
Nothing, doesn't work, #define M114_DETAIL & #define REPORT_FAN_CHANGE activated but still No printer attached is shown on the screen

kisslorand, 2021-02-03:
I tested MKS_Gen_L_V1.0 (latest bugfix2.0.x) with BT_TFT3.5_V3.0 (latest master) and it is OK. The TFT was attached to the AUX1 connector.
I attach the config files I used.
[Configurations.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5915207/Configurations.zip)

Silverender, 2021-02-03:
> I tested MKS_Gen_L_V1.0 (latest bugfix2.0.x) with BT_TFT3.5_V3.0 (latest master) and it is OK. The TFT was attached to the AUX1 connector, USB not connected.
> I attach the config files I used for Marlin.
> [Configurations.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5915207/Configurations.zip)

Thanks kisslorand will try, could you post me also your screen firmware?

kisslorand, 2021-02-03:
Just take the precompiled one: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/raw/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT35_V3.0.26.x.bin

Silverender, 2021-02-03:
> I tested MKS_Gen_L_V1.0 (latest bugfix2.0.x) with BT_TFT3.5_V3.0 (latest master) and it is OK. The TFT was attached to the AUX1 connector, USB not connected.
> I attach the config files I used for [...]

I see your machine is not an Ender 3, and also surprises me that in your CONFIG line "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" necessary for the screen is commented :-/

kisslorand, 2021-02-03:
It's not a machine, just an MKS GenL v1.0 motherboard and a BTT TFT35 v3.0  connected together for testing your problem.

Silverender, 2021-02-09:
nothing, already ordered a BTT SKR 1.4 Turbo, but will keep trying to fix it.

oldman4U, 2021-02-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

joemama626, 2021-03-01:
Oldman4U it would be nice if you were as concerned about solving problems as you are about closing out posts.  Part of the reason for posts is because more than one person might have the same issue and instead of having to resolve the issue with the community, people can better help each other with problems that haven't been solved.  Why are you so concerned about open posts?  Obviously this is still an issue for people.

oldman4U, 2021-03-01:
Hi.

Two reasons

First. To keep the ticket system here usable.

Second. This is about improving BTT firmware and not a user group. There
are other places (which are mentioned when you start a ticket) to share
user related problems.

Thank you for your understanding

oldman4U, 2021-03-01:
And in this particular case it is double wrong because it is not TFT
firmware related and the problem is most likely MKS related. So the MKS
forum would be the right place.

Georg Deisting <georgdeisting@gmail.com> schrieb am Mo. 1. März 2021 um
06:27:

VikyTT, 2021-04-13:
Hello
I'm sorry, I'm new here

I had the same problem, the display did not connect to the SKR1.4 Turbo

It was enough to change the value of the Serial port in the Config file in the oro display firmware directory
(valid for TFT35 V3)

In VisualStudioCode is coment

#### Serial Always ON
# Keep UART (serial communication) alive in Marlin Mode.
# Allows seamless OctoPrint UART connection to the TFT's UART/serial expansion port no matter which mode the TFT is in.
#   Options: [enable: 1, disable: 0]
serial_always_on:1

Of course, the baud rate of the printer and the display must also be set to the same value

Maybe I helped

Silverender, 2021-04-13:
Had to replace motherboard and screen, now working perfectly with skr 1.4
turbo and a new tft35 3.0

oldman4U, 2021-04-15:
@Msq001

This ticket is MKS mainboard related and finally has been solved by replacing the MKS with a SKR board. Please close it.

Thank you

closed 2021-04-16 by bigtreetech