bigtreetech/BIGTREETECH-TouchScreenFirmware #1579: [Q] Unable to get Marlin mode to load

doug-scott, 2021-01-29:
I have a printer running MKS SGEN_L V1.0, MKS TFT2208 in UART mode, BTT TFT35E3V30, in a cube printer.  The touchscreen is connected correctly with both ribbon cables connected in reverse on the mainboard, and the serial cable plugged into the AUX1 connection using the top row of pins(row on mainboard side of connector).
After starting printer the TFT35 only shows "LCD12864 Simulator" on the top row of the display.  The touch interface works without error.  I can do everything on the touch interface, just do not see the Marlin interface other than the title.  I have set the config.ini file to use black background, and yellow foreground in the Marlin display.  I have also copied the default config.ini file to the SD card to be sure it is not something stupid like same colour back and foreground.
I am running Marlin 2.0.7.2 firmware, and 3.0.26.x TFT35 firmware that has todays build date(downloaded current version and built firmware).

radek8, 2021-01-29:
send Marlin configuration files
Configuration.h
Configuration_adv.h

Triangulix, 2021-01-30:
I maybe have the same issue with SKR 1.4. In my case I was able to narrow down the problem to the following:
If the TFT boots up in touch mode the issue doesn't occur. But if it boots up in Marlin mode the boot process stops somewhere before the printer connection is established. Switching to touch mode finishes the boot process and the printer gets detected. After that I can switch back to Marlin mode and after clicking the encoder for a refesh Marlin mode works.

oldman4U, 2021-01-30:
@Triangulix you are not using the latest version of the firmware. This issue was fixed 3 weeks ago.

doug-scott, 2021-01-30:
I don't understand what changed, but I went through the configuration.h file and found that the line that enabled the TFT35E3V30 was commented out.  I used the online tool for editing the configuration.h file and know that I chose the correct touchscreen and saved the file before exporting it.  It now works.  One thing that I have also noticed is that it will not show the Marlin mode until you move the encoder wheel.

closed 2021-01-30 by doug-scott

radek8, 2021-01-30:
One thing that I have also noticed is that it will not show the Marlin mode until you move the encoder wheel.

This is standard behavior.
Marlin does not update the screen all the time, but only when the values change or their appearance.