bigtreetech/BIGTREETECH-TouchScreenFirmware #1581: [BUG] Biqu B1: SKR 1.4 doesn't start up if TFT is in 12864 emulation mode on power off

Drachenkaetzchen, 2021-01-30:
### Description

SKR 1.4 doesn't start up if the TFT is in 12864 emulation mode when powering off. To make SKR 1.4 start, I have to switch the LCD to touch mode, only then SKR 1.4 starts up.

### Steps to reproduce

1. Switch the TFT to 12864 emulation mode
2. Power off the printer
3. Power on the printer

**Expected behavior**
SKR 1.4 starts up, and shows a Marlin menu

**Actual behavior**
I see "12864 Simulation" on top of the screen, but nothing more happens. Marlin doesn't start up, and the typical RGB LED cycle isn't there.

To fix this, I have to switch to the touch mode, only then Marlin starts up and the RGB LED cycle happens.

### Hardware Variant
Standard Biqu B1 `TFT35` with `SKR 1.4`

### TFT Firmware Version & Main Board Firmware details
TFT Firmware: `TFT35_E3_V3.0.26.x Jan 21 2021`
Marlin Firmware: `Marlin 2.0.7.2`

### Additional Information
`config.ini` from this repo, no changes.
[Configs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5898052/Configs.zip)

BrendaIT, 2021-01-30:
Ho il solito problema....
Hardware Variant
Standard Alfawise u30 TFT35 v 3.0 with SKR 1.4 turbo

TFT Firmware Version & Main Board Firmware details
TFT Firmware: TFT35 V3.0.26.x Jan 21 2021
Marlin Firmware: Marlin 2.0.7.2

radek8, 2021-01-31:
Update FW TFT

radek8, 2021-01-31:
You say you have a Biqu B1,
but are you using FW TFT35_E3_V3.0.26.x Jan 21 2021?
You should use FW BIGTREE_TFT35_V3.0_B1.26.x.bin

BrendaIT, 2021-01-31:
Problems solved after the last update. Thanks!

Drachenkaetzchen, 2021-01-31:
> You say you have a Biqu B1,
> but are you using FW TFT35_E3_V3.0.26.x Jan 21 2021?
> You should use FW BIGTREE_TFT35_V3.0_B1.26.x.bin

I assumed the bootloader would be smart enough to pick the correct firmware when inserting the SD card, which it apparently did not.

I updated to `BIGTREE_TFT35_V3.0_B1.26.x Jan 31 2021`, the issue remains.

Drachenkaetzchen, 2021-01-31:
If I set `serial_always_on` to `1`, it seems to work. The description in the config file made no sense to me, as OctoPrint is always connected via USB to the SKR 1.4. I have no idea if this has any side effects.

kisslorand, 2021-01-31:
On the SKR 1.4 the TFT is connected to a different serial port than the USB, so no conflict.

radek8, 2021-01-31:
Screen freezing problem with serial_always_on parameter set to 1
has already been addressed and has been resolved.
Don't have an old FW for TFT?

Drachenkaetzchen, 2021-01-31:
> Screen freezing problem with serial_always_on parameter set to 1
> has already been addressed and has been resolved.
> Don't have an old FW for TFT?

I only set `serial_always_on` to `1` with the firmware `BIGTREE_TFT35_V3.0_B1.26.x Jan 31 2021`. I did not set it before.

This is most likely a documentation issue, please advise if you wish to keep this issue open to document this for B1 users or if you want me to close the issue.

oldman4U, 2021-01-31:
What exactly is the documentation issue ?

Drachenkaetzchen, 2021-01-31:
> What exactly is the documentation issue ?

That `serial_always_on: 0` causes the SKR 1.4 not to boot if 12864 mode is active on boot. That the TFT35 B1 does take the E3 firmware instead of the B1 firmware if all firmware files are copied to the sd card. That the TFT is connected to a different port than USB on the SKR 1.4 so that `serial_always_on:1` could be safely set in that scenario. That the documentation for `serial_always_on` in the `config.ini` file is hard to interpret what it exactly does.

oldman4U, 2021-01-31:
Do you want me that I explain to you how editing the ReadMe works? I do not have the B1 so I can not verify what you write.

closed 2021-01-31 by Drachenkaetzchen

radek8, 2021-01-31:
> That `serial_always_on: 0` causes the SKR 1.4 not to boot if 12864 mode is active on boot. That the TFT35 B1 does take the E3 firmware instead of the B1 firmware if all firmware files are copied to [...]

For your information
serial_always_on:1 In the past, it caused the display to freeze on some configurations when started in Marlin mode. The problem was solved.
A new type of TFT35B1 display has now been released, and you may need to test it for this problem.

When updating the display, you only need to load the FW for your display on the card.
Uploading all FWs to all models on the SD card is silly. The display reaches first in the order on the card.

Yes, the TFT is connected to a port other than USB on the SKR, but that's probably clear.
 
The value serial_always_on: 1 enables serial communication of the display while in Simulated Marlin mode. This allows you, for example, to communicate with the printer via a WIFI module connected to the TFT display in Marlin mode.

Realize that everyone who works on this FW and tries to help you does so in their free time and completely free of charge.
Have a nice day

nurplit, 2021-02-01:
Just an FYI same problem exist on TFT70 with the SKR1.4 Turbo, and the same resolve works as well. 

Thanks Dev's...

radek8, 2021-02-01:
@nurplit 
Your display will freeze when active
serial_always_on: 1
and restart in Marlin Mode?

nurplit, 2021-02-01:
@radek 
Sorry I probably didn't word that very well. 

I experienced the same thing as the original post. When the printer has been reset or powered cycled, when it was shut down in Marlin Mode (12864 emulation mode)  The printer will hang upon resetting or power cycling, with the display only showing 12864 emulation mode, and no LED start up animation will occur. As soon as the display is put into touch mode,  the LED's run there animation and the serial connection is made showing status of Ready while in touch mode. When changing the config.ini `serial_always_on:0` to `serial_always_on:1` it works as it should, allowing a restart or power cycle from Marlin Mode into Marlin mode.