bigtreetech/BIGTREETECH-TouchScreenFirmware #1582: [BUG] "Loading..." screen after canceling a print from onboard SD

Guilouz, 2021-01-30:
### Description

After canceling a print from onboard SD, TFT stay on "Loading..." screen. Need to reset screen to go back on home menu.

### Steps to reproduce

1. Start a print from onboard SD
2. Cancel this print
3. TFT stay on "Loadind..."

**Expected behavior**
After canceling a print, onboard SD card content need to be reloaded and displayed.

**Actual behavior**
After canceling a print, TFT stay on "Loading..." screen.

### Hardware Variant
TFT43 V3.0

### TFT Firmware Version & Main Board Firmware details
Latest master for TFT firmware and latest Marlin 2.0 bugfix for Main Board.

### Additional Information
No issue when canceling a print from TFT SD/USB.

kisslorand, 2021-01-30:
What motherboard do you have? Can you share a serial log when this happens?

Guilouz, 2021-01-30:
> What motherboard do you have? Can you share a serial log when this happens?

Motherboard stock on CR10S Pro (Creality 2.4 Atmega2560) and SKR 1.4 Turbo on Sidewinder X1.

By serial log you mean result in Terminal ?

kisslorand, 2021-01-30:
You have this issue on both printers? 
Serial log can be from Pronterface, Octoprint, ESP3D, or any other serial monitor.
I also have SKR1.4T on my Sidewinder X1 but I cannot reproduce the issue.

Guilouz, 2021-01-30:
> You have this issue on both printers?
> Serial log can be from Pronterface, Octoprint, ESP3D, or any other serial monitor.
> I also have SKR1.4T on my Sidewinder X1 but I cannot reproduce the issue.

Yes same issue on both.

How can use serial log with Octoprint if I print from SD Card ?

kisslorand, 2021-01-30:
Octoprint -> Connect -> Terminal

Guilouz, 2021-01-30:
This is what I have in terminal from the launch of the print until cancellation

[logs.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5898472/logs.txt)

kisslorand, 2021-01-30:
No useful info thee :(
Can you share the gCode used so I can test on my ASWX1?

Guilouz, 2021-01-30:
> No useful info thee :(
> Can you share the gCode used so I can test on my ASWX1?

[Bed_Level_Test.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5898490/Bed_Level_Test.gcode.zip)

kisslorand, 2021-01-30:
Bug confirmed.

Guilouz, 2021-01-30:
> Bug confirmed.

Do you use latest 2.x bugfix or 2.7.0.2 ?

kisslorand, 2021-01-30:
I am using Marlin bugfix 2.0.x from yesterday (29.I.2021) and latest TFT FW from today (30.I.2021).

Guilouz, 2021-01-30:
> I am using Marlin bugfix 2.0.x from yesterday (29.I.2021) and latest TFT FW from today (30.I.2021).
> I also find how to reproduce it. I hope I can come to the bottom of it tonight as I was already working on a PR with bugfixes.

So for you it's an issue with TFT firmware not with Marlin ?

oldman4U, 2021-01-30:
Try to activate the End gCode.

kisslorand, 2021-01-30:
> So for you it's an issue with TFT firmware not with Marlin ?

It's the TFT for sure.

Guilouz, 2021-01-30:
> > So for you it's an issue with TFT firmware not with Marlin ?
> 
> It's the TFT for sure.

Ok, if your need some tests, tell me.

oldman4U, 2021-01-30:
I see the issue you describe when End gCode is disabled. Enabling End gCode solves the problem when printing from on board sd card.

Guilouz, 2021-01-30:
> Try to activate the End gCode.

With ou without, same issue.

kisslorand, 2021-01-30:
It is harder than I thought. If I print a file with relative extrusion, everything is fine. As soon as I print with absolute extrusion the bug shows itself. 
Can you test with a file generated with relative extrusion to confirm my findings?

Guilouz, 2021-01-30:
> It is harder than I thought. If I print a file with relative extrusion, everything is fine. As soon as I print with absolute extrusion the bug shows itself.
> Can you test with a file generated with relative extrusion to confirm my findings?

Is it possible with Cura to define relative extrusion ? 
Or send me a gcode to test.

kisslorand, 2021-01-30:
Yes, possible in Cura. Search for "Relative" in Cura.

Guilouz, 2021-01-30:
> It is harder than I thought. If I print a file with relative extrusion, everything is fine. As soon as I print with absolute extrusion the bug shows itself.
> Can you test with a file generated with relative extrusion to confirm my findings?

Same issue with relative extrusion.

kisslorand, 2021-01-30:
Yeah, I traced it back to Marlin, still not found a workaround for it. 1:28AM here so I am going to bed now. Tomorrow is another day...

radek8, 2021-01-31:
I have another problem with interrupting printing from the integrated SD card.
FW for TFT from yesterday.
If I interrupt printing from the integrated SD card,
The printer executes G28, moves to another location, and crashes into bed.
The "M0 stop" popup window appears on the display, which cannot be confirmed.
If something were already printed, the G28 command itself would be fatal.
my CancleGcode is OK:
cancel_gcode: M104 T1 S0 \ nM104 T0 S0 \ nM140 S0 \ nG28 XY R10 \ nM107 \ nM280 P1 S90 \ nM18 \ n
Interrupting printing from a sd TFT card is OK.

[image]

kisslorand, 2021-01-31:
What if you disable your CanceGcode? Same behaviour?

radek8, 2021-01-31:
Exactly the same behavior even with off
Cance Gcode and End Gcode

kisslorand, 2021-01-31:
BTW, I have a question for all:
Why does the TFT has to return to file selection after a Cancel action? Wouldn't it be more logical to return to the main screen?

radek8, 2021-01-31:
no need to return to file selection

radek8, 2021-01-31:
https://www.youtube.com/watch?v=f4Xydj69DDM

kisslorand, 2021-01-31:
What if you cancel after all heating is done?

radek8, 2021-01-31:
I take everything back. My fault
I wanted to simulate Guilouz's mistake.
I don't normally use the obboard card and I used a file that was there a long time ago for testing. It was designed to level the bed.
When he canceled the printing, Marlin only executed the orders he had in his cache ...
The behavior is therefore standard and as expected.
I apologize for the delay.
I did not simulate the problem that Guilouz has.

kisslorand, 2021-01-31:
__Problem fixed.__

_a PR will follow ASAP_

Guilouz, 2021-01-31:
Also another issue with onboard SD and power loss recovery function. 

The PLR ​​file is correctly created at he root of SD Card during the print but after switching off and on again, the resumption of printing is not requested when returning to the internal SD. Normal ?

Working with TFT SD / USB.

kisslorand, 2021-01-31:
As far as I remember PLR is not supported for onboard SD.

Guilouz, 2021-01-31:
> As far as I remember PLR is not supported for onboard SD.

That is what I thought...

I have a mistake with this function that I can't solve, translations are not displayed correctly

In English, title, confirm and cancel buttons ok

[image]

But when I use a translation file, french here, title and confirm buttons strings are replaced by cancel string

[image]

In PowerFailed.c code looks good and my translation too : 

```
 if(mountFS()==true && powerFailedExist())
  {
    popupDrawPage(DIALOG_TYPE_QUESTION, bottomDoubleBtn, textSelect(LABEL_POWER_FAILED), (u8* )infoFile.title,
                    textSelect(LABEL_CONFIRM), textSelect(LABEL_CANCEL));
```

Guilouz, 2021-01-31:
@kisslorand Any idea ?

kisslorand, 2021-01-31:
I checked it but no idea for the moment.

PR #1588 (2021-02-02): Bugfixes, cleanup
### Requirements

BTT TFT or MKS TFT

### Description

* fixed "Loading..." screen hanging after canceling a print from onboard SD
* fixed "Loading..." screen hanging after pausing and than later canceling a print from onboard SD
* fixed nozzle moving away from parking position after "Pause" when `PARK_HEAD_ON_PAUSE` is enabled in Marlin
* fixed some memory allocation not being released after no longer needed
* fixed `request_M0` not being parsed
* fixed a hidden bug that could become active in a future development (`requestCommandInfo` is used ONLY when printing from TFT and [... cut]
closed 2021-02-02 by bigtreetech