bigtreetech/BIGTREETECH-TouchScreenFirmware #1583: [BUG] BIQU B1 - Temperature Dosen't raise...

ETE-Design, 2021-01-30:
Have just updated and BIQU B1 with the firmware 17 Hour old... I can do PID and All temperatures raise as it should and the TFT update the temp, but as soon I try to print anything the temp just stay at the last temp aviable before print... After that nothing happend cause the TFT Dosen't update the temp so the printer won't start printing... If I shut down the printer and turn it on again it show the temperature it reached just before shutdown... Seems like a BUG in the firmware cause if I revert back to stock TFT Firmware it work fine :-)

oldman4U, 2021-02-02:
Try to find the PR which broke it and call for the developer.

oldman4U, 2021-02-07:
Any news? Have you found the relevant PR?

ETE-Design, 2021-02-07:
@oldman4U Seems like it only work with their "Official" One for the B1 :-(

kisslorand, 2021-02-07:
Check if your Marlin config fulfills the reqeuested config for the TFT firmware.

oldman4U, 2021-02-07:
And maybe start a ticket in the B1 repository. For us, without owning such a printer, this is all guesswork.

gordo3di, 2021-02-09:
@oldman4U I agree

oldman4U, 2021-02-09:
Hi gordo!

Which mainboard do you have and do you have problems to Purge More when using the (Un)Load menu?

gordo3di, 2021-02-09:
Skr V1.4 Turbo, TFT 35 V3, marlin 2.6.0.1

Unload works fine. Load keeps popping up with "pause wait for user". Never got to purge more even after extruding stops just same message. Funny, I never tried "Loading" I've always just extruded via the move menu our the custom gcode commands.

closed 2021-02-10 by ETE-Design