bigtreetech/BIGTREETECH-TouchScreenFirmware #1586: [Q] What does "All LOW" error message mean?

doug-scott, 2021-01-31:
I have a MKS SGEN_L V1.0, MKS TMC2208 in UART mode, running Marlin 2.0.7.2, 3Dprobe,  BTT TFT35E3V3.0 touchscreen, with firmware version TFT35_E3_V3.0.26x Jan 29 2021 and get a pop up message on startup that simply says "All LOW".  What is it referring to, and what does it expect to see?

I am also having an issue with it not being able to home the z-axis, it just home x, then y then goes to centre of bed and deploys the probe and waits a few seconds for the bed to move, but it just stays put.  The probe will cycle 3 or 4 times before failing with the probe fail message and suggesting to fix and reset with M999.  This error is fatal, printing fails to start without having z-axis homing.  There is likely something very simple causing this, just cannot seem to locate it.  I am hoping that the for me usual thing to happen is to post a request for help, then finding the solution 10 minutes later on my own.  I tried writing this but not posting it earlier today but that did not work.

radek8, 2021-01-31:
Look at this, maybe it will help you

https://github.com/MarlinFirmware/Marlin/issues/15939

oldman4U, 2021-01-31:
Please help the community and close the ticket once you do not need it anymore. Thank you

doug-scott, 2021-02-01:
> Look at this, maybe it will help you
> 
> [MarlinFirmware/Marlin#15939](https://github.com/MarlinFirmware/Marlin/issues/15939)

I found that if you connect to your printer via the USB port with the printer not powered up you will get this error on the BTT TFT35E0V3.0 display.  When you start the printer before connecting the USB, the error is not presented.  In my situation I was reflashing the firmware from the PC to the printer's sd card via the USB cable, and this error came up after restarting the printer by disconnecting and reconnecting the USB cable.  Looks like there is some command being run on powering up that fails when only 5v power is supplied.  M122 shows the ALL LOW message when only on USB power.
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

Apply full voltage and it changes to 
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

That part is handled now, but, I still am unable to get Z axis to Home.

doug-scott, 2021-02-21:
The original topic (should have limited it to one issue/question per ticket) is resolved in so far aa I am concerned so I have closed it.

closed 2021-02-21 by doug-scott