bigtreetech/BIGTREETECH-TouchScreenFirmware #1587: Long menu labels

AntoszHUN, 2021-01-31:
Hi! Could it be that the labels next to the toggle button in the menu of certain settings that hang on the buttons might appear as a conveyor belt label?

I attach a picture of what I am talking about.

[image]

There are several such captions that hang on the buttons.
Can they run like "No printer connected" on the main screen?

Made By.: AntoszHUN

radek8, 2021-01-31:
Not necessary.
Yesterday I played with it in the CZ translation
I had to shorten the description to fit the display while being understandable and concise.

AntoszHUN, 2021-01-31:
I understand you. But you can't describe everything clearly in Hungarian and even it should be concise. Either understandable or I will abbreviate.

AntoszHUN, 2021-01-31:
The Hungarian language is slowly consisting of only abbreviations.

closed 2021-04-08 by stale[bot]