bigtreetech/BIGTREETECH-TouchScreenFirmware #1590: HOST_ACTION_COMMANDS popup Pause reboot

RepairSyl, 2021-01-31:
### Description
So on printing TFTmode with sdonboard, when i clic on pause the head and bed first go to first X0 Y0 and go back at last position of  printing, the z axe raise up 
no popup com
i clic on resume i made purge on popup pause coming wiht purge more or continue if i clic on purge more it made reboot.
when i clic on PurgeMore, it made reboot 

[image]
[Marlin-2.0.7.2-SKRminiE3v12.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5899980/Marlin-2.0.7.2-SKRminiE3v12.zip)
https://youtu.be/dAdItduYQOQ

oldman4U, 2021-01-31:
Printing from sd on board is a nightmare at the moment, like to purge filament.

This should fix the pause issue, even it purged above the print: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1580

For me, the last firmware which did not crash while purging filament (also while not pausing. print) is this:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/1eb47368ea26ee08b69ca7668ac8d4b55b0409e4

The next PR causes a crash on my system whenever I try to purge some filament: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/546bc9ac2afd81a9d3f7534a115947527efb44aa

oldman4U, 2021-02-08:
I have a several hours print going on on my machine, but I believe there could be a Marlin fix for the crash we see when the Purge More button is pressed. Here is the version which includes probably the fix for those who would like to check: 

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

sarvenn, 2021-02-13:
I tried the latest self compiled bugfix 2 firmware with all the features enabled mentioned in the config.ini file of the TFT.
I used (un)load menu. After loading, clicking on purging more still triggers a reboot.
In marlin mode, everything is OK.

oldman4U, 2021-02-13:
You have STM32 based mainboard, right?

sarvenn, 2021-02-13:
SKR Mini E3 V2 + TFT35 E3,so yes it is an STM32 @oldman4U. Your theory is right I guess (stm32 vs LPCxxx).
The thing that I could not understand is why this issue only happens in TFT mode, not in marlin mode.

oldman4U, 2021-02-13:
Because the STM32 has a problem with certain gCode commands. Try to send
M701 followed by M876 S0 via USB or other serial connection and it will
also crash. Verified with my SKR Pro v1.1.

kisslorand, 2021-02-13:
TFT uses RS232 to communicate with the motherboard and marlin mode uses another one (IIRC it's SPI).

sarvenn, 2021-02-13:
If it also crashed when sending these command via a terminal connected from USB, then shouldn't it also crash in marlin mode when doing a filament change and selecting purge more?

kisslorand, 2021-02-13:
From USB it is also RS232.

RepairSyl, 2021-02-22:
Hello, thanks all for your reply 
@oldman4U thanks for your marlin with fix, I try to find time to test it and back here to tell if it ok or not

oldman4U, 2021-02-23:
It is not fixed and it is Marlin or STM32 library related.

sarvenn, 2021-02-23:
@oldman4U hi,
Hope everything is fine.
Could you raise a ticket and take attention of the Marlin developers for this issue?
I'm inexperienced and may not answer potential questions about the issue. So I think, an experienced person from this repository is better to discuss the issue with the marlin side.
I've checked marlin repository tickets and PRs, and found these topics below, but I can not understand whether they are related or not and if they fixe the issue we had or not.
https://github.com/MarlinFirmware/Marlin/pull/21011 (you had written here)
https://github.com/MarlinFirmware/Marlin/issues/19623
https://github.com/MarlinFirmware/Marlin/pull/20928

oldman4U, 2021-02-23:
> MarlinFirmware/Marlin#21011
> MarlinFirmware/Marlin#19623
> MarlinFirmware/Marlin#20928

Hi.

I am waiting for an outstanding answer from one of the developers. But I
take care. No worries.

oldman4U, 2021-02-27:
The issue is likely fixed thanks to user arjanmels

https://github.com/MarlinFirmware/Marlin/pull/21210

Please update Marlin and close the ticket in case you do not need it anymore.

oldman4U, 2021-03-07:
This issue is fixed using the Marlin update.

Please help the community and close this ticket.

Thank you

closed 2021-03-07 by RepairSyl