bigtreetech/BIGTREETECH-TouchScreenFirmware #1603: [BUG] abort printing from mainboard SD - busy processing, print starts nevertheless

DaStivi, 2021-02-03:
### Description
i did noticed when starting ah print from mainboard SD card, and then trying to abort it, TFT displays "busy processing, please wait..."  and even the temps are not reached, it just starts homing, abl/ubl and also starts printing!

also somehow the "same" when not aborting the job, instead changing nozzle temp, in the first place it gets ignored (while still heating up), if temps reached at some point, rechange the temp and it starts todo the change... but afterwards it again looks like it does not react to another change instantly (like with TFT control) 

### Steps to reproduce

1. start print from mainboard SD
2. abort print

**Expected behavior**
print job should abort, without starting too print

**Actual behavior**
print job "continues", and even if temp is not reached it instantly starts homeing, probing and printing and not waiting for temps

### Hardware Variant
TFT35v3 B1
### TFT Firmware Version & Main Board Firmware details
latest bugfix version (+stop polling fix)

### Additional Information

DaStivi, 2021-02-03:
Something other odd I noticed right now... When Startin ah print from marlin mode, after the last print confirmation, display jumps to normal tft print screen, but don't react to touch anymore.. Still time continues too count  have reset it then. 

Regarding the first issue where the print "starts" from onboardsd after aborting... It homes, it does abl/UBL then printed first skirt, but extruder didn't even move, and then travels to 2nd skirt and then stops and homes back

oldman4U, 2021-02-12:
For the second issue you can try deactivate serial always on in config.ini

DaStivi, 2021-02-12:
> For the second issue you can try deactivate serial always on in config.ini

ok i will try this.... i did test and change quite some things in firmware, because i tried to use the BTT UPS... but i gotonly issues... with tft/printer freezing etc...

kisslorand, 2021-02-13:
For the first issue it is jut the nature of the beast. When you start the print, the buffer of the mainboard gets filled with gCodes from your gCode file. When you press "Cancel" that command gets also into the queue, it has to wait in line until the previous gCodes gets extecuted.
The  "busy processing, please wait..." message pops up and stays there as long as there is gCode command preceding the "Cancel" gCode (M24) issued when you pressed the "Cancel" button and confirmed it.

DaStivi, 2021-02-13:
> For the first issue it is jut the nature of the beast. When you start the print, the buffer of the mainboard gets filled with gCodes from your gCode file. When you press "Cancel" that command gets also into the queue, it has to wait in line until the previous gCodes are extecuted.
> The "busy processing, please wait..." message pops up and stays there as long as there is gCode command preceding the "Cancel" gCode (M24) issued when you pressed the "Cancel" button and confirmed it.

yeah i understand the why... but i don't get it, why its not like this when using TFT SD or uDisk?

kisslorand, 2021-02-13:
Because in those situation the TFT takes care of Pause, Cancel, nozzle parking, etc., not Marlin. When printing from onboard SD the TFT just sends to Marlin what the user wants and that's it, from there Marlin makes the decisions.

oldman4U, 2021-02-14:
Whats about changing serial always on?

DaStivi, 2021-02-21:
> Whats about changing serial always on?

sorry did take some time... but i've tested with serial_always_on off again... and indeed the print screen does stay into marlin mode... bad thing is now that with ESP01 you need the serial always on, to be able to access the marlin mode form the ESP Wifi Module... 🤦‍♂️🙈

oldman4U, 2021-02-22:
Now you can choose. If I remember right there has been a version which did not switch even serial always on was activated. You will have to try snd find it.

oldman4U, 2021-02-22:
I believe it was a fix from Gurmeet around 2 months ago.

oldman4U, 2021-02-27:
Could you please check https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1601 and close it.

It seems to be fixed.

Thank you

closed 2021-03-02 by DaStivi

oldman4U, 2021-03-07:
@DaStivi 

I have been searching for this ticket, because there was one not really related but somehow related, but could not remember that it was you. You already know that Gurmeet probably solved the Marlin Emulation issues with his PR: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware.git

Happy printing;-)