bigtreetech/BIGTREETECH-TouchScreenFirmware #1605: [FR] Information on the filament consumed in two-color printing

radek8, 2021-02-03:
Please complete the final information on filament consumption in two-color printing.
The values could be side by side or as a sum.

Cura generates:

M118 P0 filament_data L: [0.58, 0.24] m
M118 P0 filament_data W: [1.92794625, 0.7833581286621093] g
M118 P0 filament_data C: [1.15676775, 0.4700148771972656] Kc

the display shows:

[image]

Thank you

kisslorand, 2021-02-03:
Can you test please (when you have time) what happens when you do not include filament data in your gCode? I am curious what will the TFT calculate.

radek8, 2021-02-04:
If I delete the M118 information from the file.
The display calculates this

[image]

kisslorand, 2021-02-04:
Thank you. I will (try to) make the data displayed if filament info present to show the sum of filament data.

radek8, 2021-02-04:
Note that the githubu counts a dot as a comma.

M118 P0 filament_data L: [0.58, 0.29] m
M118 P0 filament_data W: [1.92794625, 0.9545545239257811] g
M118 P0 filament_data C: [1.15676775, 0.5727327143554687] Kc

[image]

kisslorand, 2021-02-04:
Is see no difference...

radek8, 2021-02-04:
I am sorry
punctuation marks are fine.
The translator confused the dot with a comma

kisslorand, 2021-02-05:
OK, I've done it, now the sum of  each data is shown. The character ";" cannot be a separator, as it means a comment but other than that anything (except "." of course).
The funny thing is that the code is now shorter with 8 bytes without any RAM addition. :)
A PR is imminent, I have to update the Readme and sleep... Actually the other way, sleep and update the Readme.

kisslorand, 2021-02-05:
Done: PR #1607

closed 2021-02-06 by radek8

oldman4U, 2021-03-07:
Hi radek8.

Could you please ask the user sogeniusio here: https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/28if he can provide the user name of the person who made the pdf: https://www.dropbox.com/s/k7tb4sck5d2h5bb/BTT%20TFT%20to%20BTT%20Mainboards%202021.pdf?dl=0

ripper blocked me

Maybe we can add this to the ReadMe - or do you think it would be too confusing for the average user?

Thank you

Oldman