bigtreetech/BIGTREETECH-TouchScreenFirmware #1606: Error:No Checksum with line number, Last Line: 0 on SD transfer from eps3d

jimhiggs, 2021-02-04:
Copying from ESP3D WebUI was failing up until yesterday when I updated to latest version of TFT Firmware 
Copying is now successful, thanks to the fix by @luc-github.
However I am still getting this error when I transfer files.
I am now getting successful SD copies from ESP3D to SD but the TFT display is still showing up the error shortly after the copy starts and then when the transfer is complete and I go into 'print' to list files on the SD it is stuck on the black loading screen and I have to reset the TFT to continue. After reset the copied file is then visible on the print files list.

### Steps to reproduce
Initiate copy on PC running ESP3d and WebUI
Error appears on TFT : Error:No Checksum with line number, Last Line: 0
Transfer continues  on WebUI
Transfer completes and copied file is visible in WEBUI SD files list
I clear the error message on TFT and go to print from SD, locked on black screen with "Loading" displayed.
Is locked in this state.
Reset TFT,  go to print from SD, Copied file is shown

**Expected behavior**
No Error as copy is successful.

**Actual behavior**
Error appears on TFT : Error:No Checksum with line number, Last Line: 0

### Hardware Variant
TFT35v2
### TFT Firmware Version & Main Board Firmware details
I am using the latest ESP3D and WebUI and TFT3.5 V2 2.0.26.x (resynced with master yesterday).
Marlin 2.0.7.2

### Additional Information
Does not give the error on very small .gco files

oldman4U, 2021-02-09:
Hi jimhiggs.

Long time no see. I add another user to this ticket who requested the ability to copy long time ago: @Wesie

Is this version 2.1, 2.1.x or 3?

oldman4U, 2021-02-09:
And please see ticket #1601.

jimhiggs, 2021-02-09:
Hi @oldman4U . Yes I read that ticket and that is how I learnt about the fix.
Sorry there was an extra 2 in my firmware stated above
info screen says firmware is  TFT35_V2.0.26.x Feb 4 2021

oldman4U, 2021-02-09:
I meant the ESP3D firmware.

jimhiggs, 2021-02-09:
Sorry. 2.1.

oldman4U, 2021-02-09:
Are you sure?

Last update 17. Sept 2020 and Jan. 5th, but only the ReadMe....

jimhiggs, 2021-02-09:
Pretty sure. I only updated it last week.
From ESP3D Status page: FW version: 2.1.1.b10 ESP8266/8586

closed 2021-04-17 by stale[bot]